TheThingsIndustries/lorawan-stack-docs #605: Check Node-RED docs

benolayinka, 2021-10-05:
#### Summary

@nejraselimovic we are getting a lot of feedback about problems with the Node Red send page guide. Can you walk through and check the commands work? -> https://www.thethingsindustries.com/docs/integrations/node-red/send/

- Followed and it doesn't work. The the function node you have provided does not produce the debug output. for example, where are you setting the key correlation_ids?

nejraselimovic, 2021-10-06:
I just followed the whole guide and built my setup from scratch and everything works as expected. The problem is most often one of the following:
- People forget that their router might be blocking this, so they need to set up port forwarding or turn off their antivirus (I actually added a note about this a while ago, but PR reviewers said it's out of the scope and we should remove that info)
- Misconfiguring MQTT in/out node setup with wrong TTS instance URL, MQTT username and password
- People don't read our documentation carefully, so for example they use `v3/app1/devices/dev1/down/push` for topic instead of `v3/app1@ttn/devices/dev1/down/push` on TTS CE

The biggest problem with NodeRED in general is that if you misconfigure something, you don't get any output in the console (TTS or NodeRED), just logs (that people often don't read). That makes it somewhat hard to debug and understand where an actual problem is (is it a wrong password, a wrong URL, a topic that doesn't exist, ..).

I don't see any big problems here, but I'll push a PR with slight improvements and close this issue

benolayinka, 2021-10-06:
> I just followed the whole guide and built my setup from scratch and everything works as expected. The problem is most often one of the following:
> 
> * People forget that their router might be blocking this, so they need to set up port forwarding or turn off their antivirus (I actually added a note about this a while ago, but PR reviewers said it's out of the scope and we should remove that info)
> * Misconfiguring MQTT in/out node setup with wrong TTS instance URL, MQTT username and password
> * People don't read our documentation carefully, so for example they use `v3/app1/devices/dev1/down/push` for topic instead of `v3/app1@ttn/devices/dev1/down/push` on TTS CE
> 
> The biggest problem with NodeRED in general is that if you misconfigure something, you don't get any output in the console (TTS or NodeRED), just logs (that people often don't read). That makes it somewhat hard to debug and understand where an actual problem is (is it a wrong password, a wrong URL, a topic that doesn't exist, ..).
> 
> Regarding this:
> 
> > Followed and it doesn't work. The the function node you have provided does not produce the debug output. for example, where are you setting the key correlation_ids?
> 
> I honestly don't understand what the user is trying to do here. We don't even use a function node for sending uplinks. If he was talking about scheduling downlinks and the function doesn't provide the debug output, it means he didn't configure the inject node properly, i.e. nothing is happening because he isn't injecting anything, or he didn't activate the debug node so he cannot see any debug logs. Everything that needs to be set, at least for a basic setup, is covered in the guide.
> 
> I don't see any big problems here, but I'll push a PR with slight improvements and close this issue. If you see something I don't, please let me know.

Got it. Thanks for going through the trouble.