TheThingsIndustries/lorawan-stack-docs #606: Improve Device Claiming documentation.

ymgupta, 2021-10-06:
#### Summary

- Add a section mentioning where device claiming can be used to transfer devices.
  -  I.e., within a TTSC tenant or between two TTSC tenants is possible. 
  - TTSC to TTS CE is not possible. Etc.
- Update the claim authentication code generation CLI command

#### Why do we need this ?
This helps users to decide where to use the `Device Claiming`.
#### What is already there? What do you see now?
Device claiming documentation. Ref: https://www.thethingsindustries.com/docs/devices/device-claiming/
CLI command to configure claiming using the CLI (the claim authentication code). But it returned an error while executing it. Ref: https://www.thethingsindustries.com/docs/devices/device-claiming/make-device-claimable/#claiming-settings
#### What is missing? What do you want to see?
1. Add a section mentioning the device claiming possibilities in a tabular format. Ref: https://www.thethingsindustries.com/docs/devices/device-claiming/
2. Updating the CLI command with the correct command.
```
$ ttn-lw-cli end-device set <app-id> <device-id> --claim-authentication-code.value ABCD \
  --claim-authentication-code.valid-from 2021-03-01T00:00:00Z \
  --claim-authentication-code.valid-from 2021-03-31T23:59:59Z
```
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No.

nejraselimovic, 2021-10-08:
> Add a section mentioning where device claiming can be used to transfer devices.
>
>   I.e., within a TTSC tenant or between two TTSC tenants is possible.
>  TTSC to TTS CE is not possible. Etc.

@KrishnaIyer can you please help with listing all possibilities?

KrishnaIyer, 2021-10-08:
> What about transferring between TTS Cloud and TTS Enterprise?

This is not possible currently. Device claiming only works in if the entities are in the same IS.