TheThingsIndustries/lorawan-stack-docs #652: Document resolutions for the downlink rejection errors in the Gateway Troubleshooting guide.

ymgupta, 2021-11-10:
#### Summary
Document resolutions for the downlink rejection errors in the UDP gateways. For example,  `TOO_LATE`, `TOO_EARLY`, `COLLISION_PACKET`, etc.
#### Why do we need this?
Adding resolutions for the downlink rejection errors to the [Gateway troubleshooting](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#im-noticing-some-errors-in-my-udp-gateways-packet-forwarder-logs-what-do-they-represent) documentation will be useful for the Users.
#### What is already there? What do you see now?
Cause to different downlink rejections at the gateway level.
Ref: https://www.thethingsindustries.com/docs/gateways/troubleshooting/#im-noticing-some-errors-in-my-udp-gateways-packet-forwarder-logs-what-do-they-represent
#### What is missing? What do you want to see?
Resolutions for the downlink rejection errors in the [Gateway Troubleshoot](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#im-noticing-some-errors-in-my-udp-gateways-packet-forwarder-logs-what-do-they-represent) guide.
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No.

nejraselimovic, 2021-11-11:
Could you please provide some input on what those resolutions are?

nejraselimovic, 2021-11-12:
@adriansmares I've been following the discussion on [this issue](https://github.com/TheThingsNetwork/lorawan-stack/issues/18), so thought you could provide some input here. Are causes of these errors known? Can a user do something on his side to avoid these errors?

adriansmares, 2021-11-12:
> @adriansmares I've been following the discussion on [this issue](https://github.com/TheThingsNetwork/lorawan-stack/issues/18), so thought you could provide some input here. Are causes of these errors known? Can a user do something on his side to avoid these errors? [This](https://github.com/Lora-net/packet_forwarder/blob/d0226eae6e7b6bbaec6117d0d2372bf17819c438/PROTOCOL.TXT#L420) doesn't say too much.

These errors occur only for the UDP packet forwarder, so my recommandation for anyone seeing `TxPower`/`TxFreq` errors in the gateway events would be to make sure that the `global_conf.json` is in sync - this would signal a configuration error. For the rest of the errors - I would just ask people to contact us, because the chances are quite high that there is a bug with us (or at least some edge case that we could cover for the UDP packet forwarder, since I don't really see any fixes being added to that project nowadays).

PR #661 (2021-11-17): doc: Add support info
#### Summary
Closes #652 

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Style Guidelines: Documentation obeys style guidelines in [CONTRIBUTING](CONT [... cut]
closed 2021-11-17 by nejraselimovic