TheThingsIndustries/lorawan-stack-docs #659: Document the data retention period of storage integration

ymgupta, 2021-11-17:
#### Summary
The Storage Integration allows storing received upstream messages in a persistent database and retrieving them at a later time. And, it would be good to document the retention period of data in storage integration.
#### Why do we need this?
Users would be informed about the storage retention period and could design their implementations accordingly.
#### What is already there? What do you see now?
Configuration, Enable/Disable, and retrieving of storage integration.
Ref: https://www.thethingsindustries.com/docs/integrations/storage/
#### What is missing? What do you want to see?
The retention period for the data stored in the storage integration of The Things Stack Cloud and The Things Stack Community Edition.

cc: @adriansmares
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2021-11-17:
@adriansmares @KrishnaIyer I saw a few discussion threads about the data rentention period, but I'm not sure what was the conclusion here.
- TTS CE - 24 hours?
- TTSC - 30 days?

KrishnaIyer, 2021-11-17:
That sounds correct to me.

PR #663 (2021-11-23): doc: Add retention period info, improve readability
#### Summary
Closes #659 

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Style Guidelines: Documentation obeys style gui [... cut]
closed 2021-11-23 by nejraselimovic