TheThingsIndustries/lorawan-stack-docs #684: Title: Document enabling & disabling of Storage Integration via the API

ymgupta, 2021-12-15:
#### Summary
Currently, there are instructions to enable and disable the Storage Integration for applications via Console and CLI. Also, we can enable & disable via the APIs. It would be helpful to document these APIs in the storage integration API documentation. 
Ref: https://www.thethingsindustries.com/docs/reference/api/storage_integration/
#### Why do we need this ?
To let users know that storage integration can also be enabled and disabled via APIs.
#### What is already there? What do you see now?
An API to retrieve the data from storage integration. 
Ref: https://www.thethingsindustries.com/docs/reference/api/storage_integration/#message:GetStoredApplicationUpRequest
#### What is missing? What do you want to see?
Documentation for Enabling & Disabling the storage integration using API.

**Example cURL commands:**

- Enabling Storage Integration for an application:
```
curl 'https://<domian address>/api/v3/as/applications/<application-id>/packages/associations/100' \
  -X 'PUT' \
  -H 'content-type: application/json' \
  -H 'authorization: Bearer <Application API Key>' \
  --data-raw '{"default":{"package_name":"storage-integration"},"field_mask":{"paths":["package_name"]}}'
```
- Disabling Storage Integration of an application:
```
curl 'https://<domian address>/api/v3/as/applications/<application-id>/packages/associations/100' \
-X 'DELETE' \
-H 'accept: application/json' \
-H 'authorization: Bearer <Application API Key>'
```
- Enabling Storage Integration for a device:
```
curl  'https://<domain-address>/api/v3/as/applications/<application-id>/devices/<device-id>/packages/associations/100'   \
-H 'accept: application/json'  \
-H 'Authorization: Bearer <Application API Key>'   \
-H 'Content-Type: application/json'   \
--data-raw '{"association": {"package_name": "storage-integration"},"field_mask": {"paths": ["package_name"]}}'
```
- Disabling Storage Integration for a device
```
curl 'https://<domain-address>/api/v3/as/applications/<application-id>/devices/<device-id>/packages/associations/100' \
-X 'DELETE' \
-H 'accept: application/json' \
-H 'Authorization: Bearer <Application API Key>'
```
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2021-12-15:
@benolayinka can you please give me an idea on how to incorporate this into our documentation? You worked on documenting the API so I don't know if I'm missing something, but as far as I can see, we don't really provide many examples for using API. If that's true, I don't see the point of having examples for random features only

benolayinka, 2021-12-22:
It looks to me like these APIs should be documented in Applications, but I don't see them. @adriansmares ?

@nejraselimovic API examples are at https://www.thethingsindustries.com/docs/getting-started/api/ if you would like to a few of these

adriansmares, 2021-12-22:
The service is https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.17/api/api.md#service-applicationpackageregistry .

benolayinka, 2021-12-22:
@nejraselimovic looks like the missing APIs are here  https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.17/api/api.md#service-applicationpackageregistry

Want to add an API example which is general for Adding and Removing packages, and use Storage Integration as an example?

PR #702 (2022-01-06): Using application package APIs
#### Summary
Closes #684 

#### Changes
Adding missing application package APIs and document how to use them in general. Link Storage Integration docs to these instructions in case someone wants to use the API.

The Application Packages section got a bit nasty with all these instructions for both CLI and API, so I reorganized info to two separate sections (Using the CLI and Using the API).

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted scre [... cut]
closed 2022-01-06 by nejraselimovic