TheThingsIndustries/lorawan-stack-docs #696: Document instructions for downloading `tti-lw-cli`(Enterprise CLI) binaries

ymgupta, 2021-12-24:
#### Summary
The Things Stack CLI can be installed using a package manager on Mac and Linux operating systems. If the User's operating system doesn't support package managers, then they need to rely on binaries.

Currently, the CLI binaries are available for *ttn-lw-cli* but not for *tti-lw-cli*.
#### Why do we need this ?
Enterprise users can download binaries if the package managers aren't supported in their operating systems.
#### What is already there? What do you see now?
- Instructions on how to download the `ttn-lw-cli` and `tti-lw-cli` via the package manager.
- Also, we have instructions on how to download the binaries for `ttn-lw-cli`.

Ref: https://www.thethingsindustries.com/docs/getting-started/cli/installing-cli/
#### What is missing? What do you want to see?
Instructions on how to download `tti-lw-cli` binaries for Enterprise users.
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

htdvisser, 2021-12-28:
Our download links can get pretty ugly, so I think we should implement some kind of "download link builder", similar to how we have our "cluster selector".

Download URLs for the CLI can be constructured as follows:

`https://{bucket}.s3.eu-west-1.amazonaws.com/{version}/lorawan-stack-cli_{version}_{os}_{arch}.{ext}`

The `bucket` can be:
- `the-things-stack-releases` for Open Source
- `the-things-enterprise-stack-releases` for Enterprise

The `version` (without `v` prefix) can come from Hugo configuration, but then we do need the entire thing, so `3.16.3` instead of just `3.16`.

The `os` can be:
- `darwin` (macOS)
- `linux`
- `windows`

The `arch` can be:
- `amd64` (64-bit Intel)
- `arm64` (64-bit ARM)
- `386` (32-bit Intel; only for `linux` and `windows`)
- `armv7`, `armv6` (32-bit ARM; only for `linux`)

The `ext` can be:
- `tar.gz` (only for `darwin` and `linux`)
- `zip` (only for `windows`)

---

So for The Things Stack v3.16.3 we'd have the following downloads:

https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_darwin_amd64.tar.gz
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_darwin_arm64.tar.gz
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_386.tar.gz
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_amd64.tar.gz
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_arm64.tar.gz
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_armv6.tar.gz
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_armv7.tar.gz
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_windows_386.zip
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_windows_amd64.zip
https://the-things-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_windows_arm64.zip
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_darwin_amd64.tar.gz
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_darwin_arm64.tar.gz
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_386.tar.gz
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_amd64.tar.gz
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_arm64.tar.gz
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_armv6.tar.gz
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_linux_armv7.tar.gz
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_windows_386.zip
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_windows_amd64.zip
https://the-things-enterprise-stack-releases.s3.eu-west-1.amazonaws.com/3.16.3/lorawan-stack-cli_3.16.3_windows_arm64.zip

nejraselimovic, 2021-12-29:
> Our download links can get pretty ugly, so I think we should implement some kind of "download link builder", similar to how we have our "cluster selector".
>
> Download URLs for the CLI can be [...]

I love this suggestion, @benolayinka could you pick this up please?

PR #718 (2022-01-13): Add enterprise + open source CLI download link builder

#### Summary

Closes #696 

Please double check the links

#### Screenshots

[image]

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [x] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIB [... cut]
closed 2022-01-13 by benolayinka