TheThingsIndustries/lorawan-stack-docs #704: Document a guide for connecting the LBS gateway to a local stack

ymgupta, 2022-01-06:
#### Summary
The Things Stack users are facing issues while connecting the LBS gateway to the TTS Open source stack running on an IP address.
#### Why do we need this?
It will be helpful to users to connect their LBS gateways to the local stack.
#### What is already there? What do you see now?
Ref: https://www.thethingsindustries.com/docs/gateways/lora-basics-station/
#### What is missing? What do you want to see?
Guide to connect LBS gateway to Open-source local stack.
#### How do you propose to document this?
Please find the relevant info to document here: https://github.com/TheThingsIndustries/lorawan-stack-support/issues/682#issuecomment-1005456371

cc: @KrishnaIyer 
#### Can you do this yourself and submit a Pull Request?
No.

KrishnaIyer, 2022-01-06:
I'll pick this up.

KrishnaIyer, 2022-03-23:
This is covered in https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/793

closed 2022-03-23 by KrishnaIyer