TheThingsIndustries/lorawan-stack-docs #710: Check Javascript Payload Formatter examples

benolayinka, 2022-01-10:
#### Summary

Received feedback that the Javascript payload formatters do not work when copy and pasted in the Console. Also, more examples would be useful.

https://www.thethingsindustries.com/docs/integrations/payload-formatters/javascript/

nejraselimovic, 2022-01-12:
> I just reproduced [this](http://www.thethingsindustries.com/docs/integrations/payload-formatters/javascript/#decode-uplink-example-the-things-node) and the output was slightly different than what's in the doc (`setup` instead of `button`), so I fixed that.
> 
> Then I tried reproducing [this](http://www.thethingsindustries.com/docs/integrations/payload-formatters/javascript/#encode-downlink-example-the-things-node) and got stuck. It seems that the output is always the same (`FF`), no matter which color index one tries to return. I wanted to test the function outside TTS environment so I tried this:
> 
> ```js
> var input = '{"data":{"color":"green"}}';
> 
> function encodeDownlink(input) {
>   var colors = ["red", "green", "blue"];
>   var parsed_input = JSON.parse(input);
>   var color = parsed_input.data.color;
>   return {
>     bytes: [colors.indexOf(color)],
>     fPort: 4,
>   };
> }
> 
> console.log(encodeDownlink(input));
> ```
> 
> and I'm getting the expected result, although I had to implement the `JSON.parse()` method in order to make this code work. I tried the same thing on TTS side without success (getting an error `errorSyntaxError: invalid character 'o' looking for beginning of value at parse (native)`), so now I'm not sure if I'm missing something related to this method or there's really an issue with this? @kschiffer any ideas?

kschiffer, 2022-02-08:
I can confirm this issue and this appears to be an issue originating from the backend. @johanstokking do you know what is going on?

I've output the input also to the warning array and it gets output as an empty string:

```javascript
function encodeDownlink(input) {
  var colors = ["red", "green", "blue"];
  return {
    bytes: [colors.indexOf(input.data.color)],
    warnings: [input.data.color],
    fPort: 4,
  }
}
```

```json
// Test payload
{
  "data": {
    "color": "blue"
  }
}
```

```json
// POST http://localhost:8080/api/v3/as/applications/arsars/devices/eui-1231231231212123/down/encode
{
   "downlink":{
      "f_port":4,
      "frm_payload":"/w==", // 0
      "decoded_payload":{
         "data":{
            "color":"blue"
         }
      },
      "decoded_payload_warnings":[
         ""
      ]
   }
}
```

johanstokking, 2022-02-10:
It looks like the encoders and decoders are messed up here.

The encoded `frm_payload` in the last snippet above (`/w==`) is `0xFF`. That can only be the result of the uplink decoder, not the downlink encoder.

kschiffer, 2022-02-10:
Ok, I get what the problem is: 

```json
{
  "data": {
    "color": "blue"
  }
}
```

like the docs suggest is not the correct structure. It's not required to include the `data` prop.

```
{ "color": "blue" }
```

Works.

nejraselimovic, 2022-02-17:
> Ok, I get what the problem is:
> 
> ```json
> {
>   "data": {
>     "color": "blue"
>   }
> }
> ```
> 
> like the docs suggest is not the correct structure. It's not required to include the `data` prop.
> 
> ```
> { "color": "blue" }
> ```
> 
> Works.

I've been trying to reproduce this, but some things are confusing me and I'd like to understand them in order to make the documentation as clear as possible. I have to highlight that I don't have a connected device at the moment, so I'm just blindly scheduling downlinks to see what's happening in the Console.

When I use the following downlink formatter:

```js
function encodeDownlink(input) {
  var colors = ["red", "green", "blue"];
  return {
    bytes: [colors.indexOf(input.data.color)],
    warnings: [input.data.color],
    fPort: 4,
  }
}
```

and schedule a downlink message from the Console with the following payload on FPort 1:

```json
{
    "color": "blue"
}
```

I get three events as a result. The first one is `as.down.data.receive` containing the following fields (I'm posting only relevant fields):

```json
"data": {
    "f_port": 1,
    "decoded_payload": {
      "color": "blue"
    }
  }
```

The second one is `as.down.data.encode.warning`, containing these fields:

```json
"data": {
    "f_port": 4,
    "frm_payload": "Ag==",
    "decoded_payload": {
      "color": "blue"
    },
    "decoded_payload_warnings": [
      "blue"
    ]
  }
```

where we can clearly see the `frm_payload` as a result of the `encodeDownlink()` function. The third event is `as.down.data.forward`, containing the same fields as in `as.down.data.receive`:

```json
"data": {
    "f_port": 1,
    "decoded_payload": {
      "color": "blue"
    }
  }
```

I understand the first and the second event, but third event is confusing me. I was expecting that the AS would be forwarding an encoded message.

Now, the second scenario. When I remove this line `warnings: [input.data.color]` from the formatter, i.e.:

```js
function encodeDownlink(input) {
  var colors = ["red", "green", "blue"];
  return {
    bytes: [colors.indexOf(input.data.color)],
    fPort: 4,
  }
}
```

and schedule the downlink with the same payload and on the same `FPort` as above, I only see two events, `as.down.data.receive` and `as.down.data.forward`. Both of those events contain the same data as in the first scenario, but neither of those events contains the `frm_payload`. Is this expected? It was not really intuitive to me that I only see the encoding output when I use `warnings`.

nejraselimovic, 2022-02-17:
I also cannot figure out how to use `decodeDownlink()` function. Do I need to put it after `encodeDownlink()`?

As far as I understand from [here](https://www.thethingsindustries.com/docs/integrations/payload-formatters/javascript/#decode-downlink-example-the-things-node), the downlink payload formatter would have both of these functions one after another, and when I schedule a downlink with the following JSON payload:

```json
{
    "color": "blue"
}
```

I would see event containing both `frm_payload` and `decoded_payload` fields, but I don't see that. I also don't see `downlink_message` field at all, I only see `data` field. Is this because I don't have a device connected?

This confuses me because, for example, when I schedule an uplink from the Console, I can see the `as.up.data.forward` event and `uplink_message` field in it.

johanstokking, 2022-02-22:
I submitted a fix for the issue that the forwarded downlink event does not contain the data that is actually being forwarded to NS, namely the encoded and encrypted downlink. See #5241. Hope that clarifies.

Regarding the downlink decoder; it's the inverse of the downlink encoder. The downlink decoder decodes decrypted FRMPayload into an object, just like the uplink decoder. The reason we have a downlink decoder is to represent the data structure if we only have the binary payload available, i.e. when a downlink related event was received from the Network Server.

> I would see event containing both `frm_payload` and `decoded_payload` fields, but I don't see that. I also don't see `downlink_message` field at all, I only see `data` field. Is this because I don't have a device connected?

Can you clarify this?

nejraselimovic, 2022-02-23:
> I submitted a fix for the issue that the forwarded downlink event does not contain the data that is actually being forwarded to NS, namely the encoded and encrypted downlink. See https://github.com/TheThingsNetwork/lorawan-stack/pull/5241. Hope that clarifies.

This will for sure eliminate some questions.

> Can you clarify this?

[This section](http://localhost:1313/integrations/payload-formatters/javascript/#decode-downlink-example-the-things-node) says that if the downlink payload formatter looks like this:

```js
function decodeDownlink(input) {
  switch (input.fPort) {
  case 4:
    return {
      data: {
        color: ["red", "green", "blue"][input.bytes[0]]
      }
    }
  default:
    throw Error("unknown FPort");
  }
}
```

and you schedule a downlink on an FPort 4 with payload as bytes `02`, the resulting downlink message would contain the following:

```json
{
  "downlink_message": {
    "f_port": 4,
    "frm_payload": "Ag==",
    "decoded_payload": {
      "color": "blue"
    }
  }
}
```

meaning the bytes from the payload (`02`) should be decoded using the function above and visible under `decoded_payload`object. The issue is that I don't see that. I see this:

```
"data": {
    "f_port": 4,
    "frm_payload": "Ag=="
  }
```

Now I'm not sure if the problem is me misinterpreting the docs (or missing something else), or the docs aren't accurate 🤷‍♀️

johanstokking, 2022-02-23:
> The issue is that I don't see that. I see this a downstream event that contains:
> 
> ```json
> "data": {
>     "f_port": 4,
>     "frm_payload": "Ag=="
>   }
> ```

The downstream event indeed doesn't run the decoder.

If you would be listing the downlink queue, or if you would be observing downlink events like downlink sent, you should see them.

I think that indeed we have a discrepancy between downlink message events that are generated by AS (like queued) and by NS (like sent), where the former aren't ran through the downlink decoder but the latter are.

> > Downlink messages sent upstream as part of events or downlink queue operations are therefore decoded, just like uplink messages.
> 
> but I don't see any upstream event in the Console when I schedule the above mentioned downlink. Is this maybe why I don't see the decoded payload anywhere?

So I think indeed the answer is yes.

> This object does contain `decoded_payload` indeed, but that field isn't a result of downlink payload decoding, because I see the same content in this event even when I completely remove the `decodeDownlink()` from payload formatter. I again don't see any upstream events though..

Right, that decoded downlink is what you scheduled, not what has been decoded from binary payload.

---

@adriansmares if binary payload is being scheduled and there is a downlink _decoder_, should we also run the downlink decoder on the binary payload, just to complete the `decoded_payload` in any upstream messages, including downlink queued events?

adriansmares, 2022-02-23:
> @adriansmares if binary payload is being scheduled and there is a downlink _decoder_, should we also run the downlink decoder on the binary payload, just to complete the `decoded_payload` in any upstream messages, including downlink queued events?

The answer is obviously yes, but isn't this happening already ? We run the downlink decoder on:
- Downlink queue list
- Downlink nack/queued/sent/failed/ack

What other call sites are we missing ? Maybe we're sending the pre-decoded message in some events.

johanstokking, 2022-02-24:
> * Downlink nack/queued/sent/failed/ack
> 
> What other call sites are we missing ? Maybe we're sending the pre-decoded message in some events.

I'm specifically talking about the downlink queued event. It looks like we have two occasions it gets sent:

1. https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.18/pkg/applicationserver/applicationserver.go#L732-L733. However, I don't think Network Server sends this event at all
2. https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.18/pkg/applicationserver/observability.go#L352-L357. This is called directly from the downlink queue operation method on a queue push or replace call. There is no code path here that decodes downlink

I think we can remove (1) and call `decodeDownlink()` on the downlink queue operation.

ACK?

adriansmares, 2022-02-24:
ACK x2.

johanstokking, 2022-02-24:
@nejraselimovic this issue now also tracks the TTS changes mentioned in https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/710#issuecomment-1049622876. I'll pick that up.

Is everything clear for your for the documentation part now?

nejraselimovic, 2022-02-24:
> @nejraselimovic this issue now also tracks the TTS changes mentioned in [#710 (comment)](https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/710#issuecomment-1049622876). I'll pick that up.
> 
> Is everything clear for your for the documentation part now?

Yes, thank you for explaining and for suggesting those changes!

johanstokking, 2022-02-25:
I decided to file an actual issue for this so I can track it in a version milestone.

PR #721 (2022-03-18): doc: Fix uplink decoder example
#### Summary
Closes #710 

#### Notes for Reviewers
Rearranged to separate docs because it was too long and contained bunch of info. I tried to explain a bit more how things work, but had issues on how to make it simple and easily readable, so any suggestions or corrections are warmly welcome.

P.S. I'm planning to update the downlink decoder part when [this](https://github.com/TheThingsNetwork/lorawan-stack/issues/5260) is done.

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the d [... cut]
closed 2022-03-18 by nejraselimovic