TheThingsIndustries/lorawan-stack-docs #714: Add note regarding TTIG claiming in various TTS distributions.

ymgupta, 2022-01-11:
#### Summary
The Things Indoor Gateway can be added to The Things Stack via a process called Gateway Claiming. Currently, users can connect their TTIGs with TTS Cloud and TTS CE deployments.

But there is no information on the possibilities of connecting TTIG to other TTS deployments such as Enterprise, AMI, AWS ECS, Dedicated Cloud, and Open-source.
#### Why do we need this?
To give clear information on connecting TTIG to TTS in various The Things Stack distributions.
#### What is already there? What do you see now?
TTIG claiming in TTS Cloud and TTS CE deployments. 
Ref: https://www.thethingsindustries.com/docs/gateways/thethingsindoorgateway/
#### What is missing? What do you want to see?
...
#### How do you propose to document this?
Currently, it is not possible to connect TTIG to other TTS distributions apart from TTS Cloud Hosted and TTS CE.

CC: @KrishnaIyer 

Add a note conveying the possibility of connecting TTIG (TTIG claiming) to various TTS distributions.
#### Can you do this yourself and submit a Pull Request?
No.

nejraselimovic, 2022-01-14:
Since it is possible to claim TTIG in other deployments after all (see discussion in #724), I will close this issue and we will document the claiming process with #728

KrishnaIyer, 2022-01-14:
@nejraselimovic: It's not possible to **claim** TTIGs in other deployments. However it's possible to **connect**.  Those are different things and the latter is what my comment refers to.

nejraselimovic, 2022-01-14:
> @nejraselimovic: It's not possible to **claim** TTIGs in other deployments. However it's possible to **connect**. Those are different things and the latter is what my comment refers to.

Aha, I got it now, those terms are mixed in the issue too, so we definitely need to clarify that. Please pick this up with the same PR as for #728

KrishnaIyer, 2022-03-23:
I think this is covered in https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/793

closed 2022-03-23 by KrishnaIyer