TheThingsIndustries/lorawan-stack-docs #715: Document the Packet Broker Mapper API and its frequently asked questions.

ymgupta, 2022-01-11:
#### Summary
We have a Packet Broker Mapper API for fetching gateway locations, online status and other info for The Things Stack Community Edition, Cloud, Open Source and Enterprise.
Ref: https://www.thethingsnetwork.org/forum/t/new-api-for-gateway-mapping-status-and-info/49778
#### Why do we need this?
Documenting Packet Broker Mapper API info will be helpful for users to fetch their gateways related information when using the Packet Broker service.
#### What is already there? What do you see now?
Ref: https://packetbroker.net/getting-started/
#### What is missing? What do you want to see?
Document the overview of the Packet Broker Mapper API and its frequently asked questions.
#### How do you propose to document this?
@nejraselimovic,
Please refer to the following community forum reference for documentation on PB Mapper API and its frequently asked questions. Ref: https://www.thethingsnetwork.org/forum/t/new-api-for-gateway-mapping-status-and-info/49778
#### Can you do this yourself and submit a Pull Request?
No.

nejraselimovic, 2022-01-27:
@benolayinka Can we configure this to be documented in the same way as TTS API (semi-automatic)? Could you pick it up by any chance or at least give me a starting point?

benolayinka, 2022-02-14:
@johanstokking do you have an idea?

johanstokking, 2022-02-22:
This documentation is probably two-fold:

1. Document Packet Broker Mapper API on https://packetbroker.net
2. Document common tasks in TTS docs

For (1), please file issue in https://github.com/packetbroker/docs if you want this. If so, I would basically go with:
- Document that we have an OpenAPI definition. Link to existing OpenAPI docs. Guide people to explore it. Steps are: 1) paste `https://petstore.swagger.io/ 2) paste https://mapper.packetbroker.net/api/v2/openapi.json` in top bar and click **Explore**. But that OpenAPI definition can be used for many other things, see https://swagger.io/specification/
- Add some examples (see below)

For (2), I would say:
- Add **Mapper API** page under https://www.thethingsindustries.com/docs/getting-started/packet-broker/
- Link to OpenAPI definition and to OpenAPI specification
- Add some examples (see below)

---

**Examples**

Get all gateways:

```
curl https://mapper.packetbroker.net/api/v2/gateways
```

Get a single gateway:

```
curl https://mapper.packetbroker.net/api/v2/gateways/netID=000013,tenantID=ttn,id=rabobank-utrecht
```

Output:

```json
{
  "netID": "000013",
  "tenantID": "ttn",
  "id": "rabobank-utrecht",
  "eui": "0000024B080E0FFD",
  "clusterID": "eu1.cloud.thethings.network",
  "updatedAt": "2022-02-22T11:52:01.520808Z",
  "online": true,
  "frequencyPlan": {
    "region": "EU_863_870",
    "loraMultiSFChannels": [
      868100000,
      868300000,
      868500000,
      867100000,
      867300000,
      867500000,
      867700000,
      867900000
    ]
  },
  "rxRate": 2821.1191,
  "txRate": 151.6022
}
```

johanstokking, 2022-03-04:
@nejraselimovic do you have enough information to pick this up?

johanstokking, 2022-06-24:
@nejraselimovic I invited you for the organization. I think you lost access when we reorganized permissions. Please see your email with an invitation link.

closed 2022-07-15 by nejraselimovic