TheThingsIndustries/lorawan-stack-docs #723: Document the "DevEUI "XXXXX" is registered for mismatching stack (0xc000868500): errorString null" error in the AWS IoT core troubleshooting guide.

ymgupta, 2022-01-13:
#### Summary
In AWS IoT core, there would be a conflict if a thing with the same DevEUI already exists but was created by a different CloudFormation stack other than the one from which uplinks are currently received for the device. Below is the error raised by the HandleUplink lambda function (can be seen in the CloudWatch logs).
```
DevEUI "xxxxxx" is registered for mismatching stack (0xc000868500): errorString null
```
#### Why do we need this?
It would be helpful for users to resolve the issues with AWS IoT Integration.
#### What is already there? What do you see now?
A troubleshooting guide that covers what CloudWatch logs to check if there are any AWS IoT integration issues.
Ref: https://www.thethingsindustries.com/docs/integrations/cloud-integrations/aws-iot/default/troubleshooting/
#### What is missing? What do you want to see?
Document the resolution to fix this issue in the AWS IoT troubleshooting guide.

> In order to fix this issue, update the `stackName` parameter of the existing thing with the CloudFormation stack name (name of the latest stack created) from which the uplinks are receiving.

cc: @johanstokking 
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

johanstokking, 2022-01-13:
Done. Docs are building, but this is what it's gonna look like:

[image]

nejraselimovic, 2022-01-14:
> Done. Docs are building, but this is what it's gonna look like:
> 
> [image]

nice, thanks for picking this up! can the issue be closed?

johanstokking, 2022-01-14:
Yes!

closed 2022-01-14 by johanstokking