TheThingsIndustries/lorawan-stack-docs #727: Improvements in AWS Marketplace AMI documentation

ymgupta, 2022-01-14:
#### Summary

Improve [The Things Stack AWS Marketplace AMI](https://www.thethingsindustries.com/docs/getting-started/aws/ami/) documentation.

#### Why do we need this ?

To assist users in understanding the upgrading process and troubleshooting the issues.

#### What is already there? What do you see now?

Instructions for [AWS Market Place AMI](https://www.thethingsindustries.com/docs/getting-started/aws/ami/) deployment, post-deployment configuration, updating, and troubleshooting guides.

#### What is missing? What do you want to see?
**[Troubleshooting AWS AMI Deployment](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/):**
1. Document the cause & resolution of the `connect: connection refused` error that would occur while performing the database schema migrations.
    - It occurs if the configuration file is not exported before executing the commands to perform the database schema migrations.
    - To resolve this issue, users can either do `export TTN_LW_CONFIG=/tti/lorawan-stack/config.yml` or for each command pass the configuration file as `-c /tti/lorawan-stack/config.yml`. Please see the commands below.
      ```
      /tti/lorawan-stack/tti-lw-stack -c /tti/lorawan-stack/config.yml is-db migrate
      /tti/lorawan-stack/tti-lw-stack -c /tti/lorawan-stack/config.yml ns-db migrate
      /tti/lorawan-stack/tti-lw-stack -c /tti/lorawan-stack/config.yml as-db migrate
      ```
    - FYI, error while performing database schema migration:
      ```
      $ /tti/lorawan-stack/tti-lw-stack is-db migrate
      WARN  No license configured, running unlicensed mode  {"component_address_regexps": ["localhost"], "dev_addr_prefixes": ["00000000/7","02000000/7"], "license_id": "unlicensed", "max_applications": 10, "max_clients": 10, "max_end_devices": 10, "max_gateways": 10, "max_organizations": 10, "max_users": 10, "valid_from": 1641918284.7304285, "valid_until": 1641921884.7304285}
      INFO  Connecting to Identity Server database...
      dial tcp 127.0.0.1:5432: connect: connection refused
      correlation_id=31856630e43747eb9c826d658fbbc23c
      ```
**[Updating the CloudFormation Stack](https://www.thethingsindustries.com/docs/getting-started/aws/ami/updating/):**
1. Add a note point to convey to the users that, any changes to AMI deployments must be made by editing the Cloud Formation template parameters only. If any change is made outside the CF template, this would result in the *Drift* which causes issues when the EC2 instance is restarted.
2. Redirect the users to download the new version of the CloudFormation template from the [AWS Markerplace](https://aws.amazon.com/marketplace/pp/prodview-okhh3ofzhqj56?qid=1593444260869&sr=0-1&ref_=srh_res_product_title).
    - `Usage` ---> `View CloudFormation Template` ---> `Download CloudFormation Template`.
3. In [Manual Upgrade of Databases (Amazon RDS and ElastiCache) using snapshots](https://www.thethingsindustries.com/docs/getting-started/aws/ami/updating/#manual-upgrade-of-databases-amazon-rds-and-elasticache-using-snapshots) section, to restore the Redis Database from a snapshot, it is documented as:

    > For Redis (ElastiCache), enter the Name of the backup into the **Amazon RDS Database Name** and run the changeset.

    The parameter should be `Amazon ElastiCache Redis Snapshot`.

4. Redirect the users to the [Troubleshooting AWS AMI Deployment](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/) section in case of any issues observed while performing the [Database Schema Migrations](https://www.thethingsindustries.com/docs/getting-started/aws/ami/updating/#database-migrations)

**[Database Migrations](https://www.thethingsindustries.com/docs/getting-started/aws/ami/updating/#database-migrations):**
1. Update the title to `Database Schema Migrations`
2. AMI users can perform database schema migrations by calling TTS as `/tti/lorawan-stack/tti-lw-stack`. Also, the configuration file should be exported before executing the commands. 

    For example, below are the steps to be performed to do IS DB migration
    ```
    export TTN_LW_CONFIG=/tti/lorawan-stack/config.yml

    /tti/lorawan-stack/tti-lw-stack is-db migrate
    ```
**[Resource Settings](https://www.thethingsindustries.com/docs/getting-started/aws/ami/deployment-guide/#security-configuration):** 
1. Add description for the parameters `Amazon ElastiCache KMS Key ID` & `Amazon ElastiCache Password`, that a change on these fields requires manual migration of the databases.

cc: @michalborkowski96 
#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2022-01-14:
I think it's better for @michalborkowski96 to take this one

michalborkowski96, 2022-01-20:
Documentation updated, changed should be picked up by the servers within an hour or two.

PR #735 (2022-01-20): Fixes to the AWS Marketplace documentation
#### Summary
Closes #727 

#### Changes
A few minor fixes

#### Checklist

- [ ] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [ ] Style Guidelines: Docum [... cut]
closed 2022-01-20 by michalborkowski96