TheThingsIndustries/lorawan-stack-docs #733: Update the command-line interface documentation

ymgupta, 2022-01-19:
#### Summary
The Enterprise CLI (`tti-lw-cli`) has additional commands for tenant management and OpenID Connect, which are not available with `ttn-lw-cli`. Currently, in the [command-line interface documentation](https://www.thethingsindustries.com/docs/reference/cli/), the enterprise CLI (`tti-lw-cli`) commands are also documented to use with the open-source CLI `ttn-lw-cli`.
#### Why do we need this ?
To distinguish the commands available for the `tti-lw-cli `and `ttn-lw-cli`.
#### What is already there? What do you see now?
The Things Stack Open source command-line interface **ttn-lw-cli** documentation includes enterprise CLI commands.
Ref: https://www.thethingsindustries.com/docs/reference/cli/
#### What is missing? What do you want to see?
A distinction should be made between open-source and enterprise CLI commands in the documentation.
#### How do you propose to document this?

1. Update the overview of the CLI docs that it is the reference for both open source and enterprise CLI.
Ref: https://www.thethingsindustries.com/docs/reference/cli/
2. Create the command tree for enterprise CLI.
#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2022-01-19:
I think this is automatically generated when updating docs after a TTS release, but I'm not so sure how that works so I'm gonna assign @benolayinka

KrishnaIyer, 2023-10-01:
I'll pick this one up myself.