TheThingsIndustries/lorawan-stack-docs #743: Add AWS ECS deployment troubleshooting guide

ymgupta, 2022-01-26:
#### Summary
Add troubleshooting guide for AWS ECS deployment with the common issues observed during installation and upgrading of ECS deployment.
#### Why do we need this?
It would be helpful for users to resolve the issues by referring to the documentation.
#### What is already there? What do you see now?
Instructions for installation and upgrading of [AWS ECS](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/) deployment.
#### What is missing? What do you want to see?
Document the below issues observed while installing and upgrading the AWS ECS deployment.
1. Note in the documentation that using `v3.X` as Docker images in the CF template will update the TTS components to the latest patch version available (`v3.X.X`) in that minor version. Hence, the users must specify the patch version explicitly as the Docker images to update to that specific version. For example, a user who wants to upgrade to `v3.11.0`, has the Docker image version set to `3.11`, and it pulled the latest Patch version`v3.11.4`.
2. We have observed Identity server store errors `error:pkg/identityserver/store:database (database error)` when logging into the console after updating The Things Stack. This could be due to a broken connection between the Identity Server and the database. So, restarting the ECS tasks of the Identity Server will resolve the issue.
3. The session information is lost for all devices, and the uplinks are dropped by the Network Server.
   - Redis will store the session information for devices communicating with the Network Server. If there is any issue/crash with the Redis database, it causes the loss of devices session data. Hence, it is always recommended to take backups of the databases as snapshots. Users can restore session information from available snapshots if they encounter any issues while updating TTS or any other issue (for example, an AWS issue).
4. AWS ECS deployment troubleshooting
     - AWS-specific issues (e.g., Continuous failures of tasks, pending tasks) are outside the scope of our support. Users can refer to [ECS Troubleshooting](https://docs.aws.amazon.com/AmazonECS/latest/developerguide/troubleshooting.html) Section for more information on the AWS ECS issues.
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

michalborkowski96, 2022-02-01:
https://www.thethingsindustries.com/docs/getting-started/aws/ecs/troubleshooting/

Is this good?

ymgupta, 2022-02-02:
Thanks, @michalborkowski96. Yes, this is good.

We are closing this issue now.

closed 2022-02-02 by ymgupta