TheThingsIndustries/lorawan-stack-docs #755: Update AWS IoT integration Metrics documentation with latest AWS IoT Core UI

ymgupta, 2022-02-01:
#### Summary
The AWS IoT integration for The Things Stack reports device metrics to the shadow state. The current screenshot of the `Shadow state` and navigation steps to the `Device Shadow State` is outdated.

Ref: https://www.thethingsindustries.com/docs/integrations/cloud-integrations/aws-iot/default/metrics/
#### Why do we need this?
To avoid confusion for users when they follow the documentation to get Device Shadow State metrics.
#### What is already there? What do you see now?
A guide to fetch the LoRaWAN shadow state metrics maintained by The Things Stack.

Ref: https://www.thethingsindustries.com/docs/integrations/cloud-integrations/aws-iot/default/metrics/
#### What is missing? What do you want to see?
- The current navigation steps are as below in the new UI (refer to the screenshot).
```
Services ---> IoT Core ---> Manage ---> Things ---> ThingName ---> Device Shadows ---> ThingTypeName
```
[image]

- Also, add details related to the `Device Shadow Metadata` (refer to the screenshot) that has a timestamp indicating when the state was last updated for each attribute in the `Device Shadow State` section.

[image]

cc: @johanstokking
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

johanstokking, 2022-02-02:
@nejraselimovic are you unable to update the screenshots or what do you need help with?

nejraselimovic, 2022-02-09:
> @nejraselimovic are you unable to update the screenshots or what do you need help with?

Yeah, I don't have access to AWS IoT Core, and even if I did, I guess I'd have to go through configuring everything to take these screenshots. I suppose you have something configured on your account already, so could you please provide screenshots? I can then create the PR. I'm also not sure if the new AWS IoT Core UI makes differences other than this one compared to our documentation

PR #784 (2022-02-23): Update AWS IoT Core to new console

#### Summary

Closes #755 

#### Screenshots

[image]

[image]

#### Changes

- Update screenshots to reflect the new console of AWS IoT Core. Also updated some texts that have changed
- Moved the deprecated AS telemetry in AWS IoT to the AMI deployment, as it's only relevant there
- Because of that, the "default integration" is now the only thing mentioned under AWS IoT. This simplifies the navigation a lot

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using ` [... cut]
closed 2022-02-23 by johanstokking