TheThingsIndustries/lorawan-stack-docs #757: Document about the Custom ADR feature

ymgupta, 2022-02-01:
#### Summary
TTS supports the Custom ADR feature to control the ADR parameters in the device manually. Document the process of implementing the Custom ADR in the TTS.

##### Why do we need this?
To help users know about the Custom ADR and the implementation procedure.

#### What is already there? What do you see now?
The ADR algorithm implemented in TTS.
Ref: https://www.thethingsindustries.com/docs/reference/adr/

#### What is missing? What do you want to see?
Document about Custom ADR and how to use that feature.

#### How do you propose to document this?

Document the below steps in the ADR documentation as a separate section.

---
**Custom ADR:**
To manually control the ADR parameters, one can disable the automatic ADR (set `use_adr` to false) and then alter the ADR parameters. Once the ADR is disabled for the device, the Network Server will no longer try to optimize the end device parameters (`Data Rate`, `Tx Power` & `NBTrans`).

Now, manually set the below ADR Parameters (SF, NBtrans, and Tx power) for the devices.

1. `adr-data-rate-index`
2. `adr-tx-power-index`
3. `adr-nb-trans`

**Implementation procedure:**
**Step 1:** Turn off ADR
**Command:**
```
ttn-lw-cli end-devices set --application-id <app-id> --device-id <dev-id> --mac-settings.use-adr=false
```
**Step 2:** 
Modify the mac-state.desired-parameters

1. `adr-data-rate-index`
2. `adr-tx-power-index`
3. `adr-nb-trans`

**Command:**
```
ttn-lw-cli end-devices set --application-id <app-id> --device-id <dev-id> --mac-state.desired-parameters.adr-data-rate-index <data_rate> --mac-state.desired-parameters.adr-tx-power-index <power_index> --mac-state.desired-parameters.adr-nb-trans <nb_trans>
```
**Note:** 

1. After executing the command, the NS will schedule **LinkADRReq** with the set parameters upon reception of the next uplink. The device should answer with **LinkADRAns**. If the device accepts all the three parameters that are specified, then only it will use them.
2. It is suggestible to set all the three-parameter values (supported by the device based on the **MAC**, **PHY** versions, and the **frequency plan**) mentioned above to avoid rejection.
3. Please test the process on test devices before you implement it on the devices in production devices.
4. The changes to `mac-state.desired-parameters.<parameter>` are not persistent and will be lost on rejoin/reset of the device.

--
#### Can you do this yourself and submit a Pull Request?
No.

ymgupta, 2022-02-19:
Hi @nejraselimovic,

1. Along with the Custom ADR, document about the ADR margin with an example.
2. Also, include information about ADR margin in the MAC Settings section, and redirect users to the section where the above information is documented.

---

Most of the users are confused to configure the ADR margin. We can provide a sample example configuration and the expected behavior for the configuration. We can include the below content in the documentation.

**Configuring ADR Margin:**
Network Server calculates the margin to further optimize the data rate, and a part of the calculation is configurable per device using the ADR Margin parameter (`--mac-settings.adr-margin`). By default, the Network Server uses the ADR Margin of **15**.

- Increasing the ADR margin on all devices tends to lower the data rate used, and therefore reduces the overall network capacity. But increases the link margin and will tend to lower the packet loss rate in lightly loaded networks.
- On the other hand, lowering the ADR margin tends to push data rates up, increases network capacity but will increase packet loss rate.

**Command**: `ttn-lw-cli end-devices set --application-id <app-id> --device-id <device-id> --mac-settings.adr-margin <float32>`
**Note:**

- It accepts a float value.
- The value is persistent after the join.

**Note**: Please make sure you test this on test devices before you update for devices in production. Also, test with different values to identify which one works better for your physical deployment.
**Example**
Consider a device for which the SNR values of the last 20 uplinks received are in the range of `0` to `7`, and the current DR is **DR3**.
**Case 1**: Consider we haven't configured the **ADR margin** (`margin_db`). Now the default value is considered as `15`.
- First, the Max (not average) SNR (SNRm) is computed for the recent uplinks we considered.
  - In our case, the `SNRm` is `7`.
- Now calculate the `SNRmargin`,
   - SNRmargin = SNRm – SNR(DR) - margin_db
   - SNRmargin = SNRm – SNR(DR3) - margin_db = 7 - (-12.5) - 15 = 4.5
- Next step is to calculate the **NStep** value.
   - NStep = int(SNRmargin/3) = int(4.5/3) = 1

Considering the **NStep** value from the calculation and the **Optimal Rate determination Algorithm** described [here](https://www.thethingsnetwork.org/forum/uploads/default/original/2X/7/7480e044aa93a54a910dab8ef0adfb5f515d14a1.pdf), the Network Server will try to increase the Data Rate to **DR4**.
**Case 2**: Consider we have configured the **ADR margin** (`margin_db`) as `17` for this device.
- **SNRm** = `7` & **margin_db** = `17`
- **SNRmargin** = SNRm – SNR(DR3) - margin_db = 7 - (-12.5) - 17 = 2.5
- **NStep** = int(SNRmargin/3) = int(2.5/3) = 0

Here, the Network Server will not try to optimize anything.
**Case 3**: Consider we have configured the **ADR margin** (`margin_db`) as `25` for this device.
- **SNRm** = `7`& **margin_db** = `25`
- **SNRmargin** = SNRm – SNR(DR3) - margin_db = 7 - (-12.5) - 25 = -5.5
- **NStep** = int(SNRmargin/3) = int(-5.5/3) = -1

In this case, the Network Server will increase the TXPower by 3dB for one time, and DR will not be altered.

---

CC: @adriansmares, please validate the ADR margin content & example.

adriansmares, 2022-02-21:
The only thing I want to add to the above is that `NStep` is calculated by dividing to `2.5`, not `3`. The `2.5` is apparently the range between two data rates in dB.

PR #798 (2022-03-04): Doc/custom adr
#### Summary
Closes #757 

#### Notes for Reviewers
I included examples noted in the issue #757, but I have trouble understanding how users can decide which value to set for ADR margin. I see that depending on the margin used, the output is different, meaning the Network Server will (or not) take an action like increasing the data rate or Tx power. But how do you decide which action is "the right one"?

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screens [... cut]
closed 2022-03-04 by nejraselimovic