TheThingsIndustries/lorawan-stack-docs #758: Document the best practices for deploying and maintaining the self-hosted deployments

ymgupta, 2022-02-02:
#### Summary
Add the best practices for deploying and maintaining the self-hosted deployments to the documentation.

#### Why do we need this?
It would be helpful for users to set up and maintain the self-hosted deployment effectively.

#### What is already there? What do you see now?
The documentation has instructions to set up self-hosted deployments in various ways (AWS AMI, AWS ECS & Docker).
Refs:
Docker deployment: https://www.thethingsindustries.com/docs/getting-started/installation/
AWS AMI deployment: https://www.thethingsindustries.com/docs/getting-started/aws/ami/
AWS ECS deployment: https://www.thethingsindustries.com/docs/getting-started/aws/ecs/

#### What is missing? What do you want to see?
The suggested best practices for self-hosted deployments. 

#### How do you propose to document this?
Below are some of the best practices we have identified so far. Please feel free to add additional points from your end (if any).
1. For High Availability of the deployment, it is recommended to set up redundancy in databases. Because the components of The Things Stack are inherently stateless, they depend on the underlying two databases - Redis and Postgres for the information. For example, you can enable Multi-AZ for Amazon RDS and Redis in AWS AMI deployment.
2. Maintain a staging/development environment to test the upgrades and perform additional tests.
3. Use managed databases such as AWS RDS, AWS ElastiCache, etc. Also, make sure to take regular backups of the databases.
4. When purchasing an Enterprise license, consider setting up a multi-tenant environment from the beginning if you plan to add tenants later. Because switching from a single-tenant to a multi-tenant environment requires a database migration.

#### Can you do this yourself and submit a Pull Request?
No, @michalborkowski96

michalborkowski96, 2022-03-16:
I have added "Best Practices" section to AWS AMI and AWS ECS deployment guides. Within an hour the cache should refresh and you should see new pages.

Regarding Docker deployment, we kinda of assume that the customer knows what they're doing.

ymgupta, 2022-03-17:
Thanks, @michalborkowski96. The best practice guides seem good.

> Regarding Docker deployment, we kinda of assume that the customer knows what they're doing.

Ok. We are closing this issue now.