TheThingsIndustries/lorawan-stack-docs #764: Update the order of API calls for adding devices in the Using API documentations.

ymgupta, 2022-02-05:
#### Summary
In the documentation, there is a mismatch in the order of adding devices in different servers specified in [Using the API](https://www.thethingsindustries.com/docs/getting-started/api/#multi-step-actions) documentation and the order specified in the [End Device APIs](https://www.thethingsindustries.com/docs/reference/api/end_device/) documentation.

#### Why do we need this?
To maintain consistency across the documentation.
#### What is already there? What do you see now?
Currently, the order specified in the [End Device APIs](https://www.thethingsindustries.com/docs/reference/api/end_device/) section is `Identity Server`, `Join Server`, `Network Server`, and `Application Server`. Whereas the order specified in the [Using the API](https://www.thethingsindustries.com/docs/getting-started/api/#multi-step-actions) section is `Identity Server`, `Network Server`, `Application Server`, and `Join Server`.

#### What is missing? What do you want to see?
Maintain the same order of execution of APIs throughout the documentation for adding a device.

#### How do you propose to document this?

1. Update the order of API calls in the [Using the API](https://www.thethingsindustries.com/docs/getting-started/api/#multi-step-actions) documentation as specified in [End Device APIs](https://www.thethingsindustries.com/docs/reference/api/end_device/) documentation.
2. Also, make sure to document the same order while documenting here: https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/747.

#### Can you do this yourself and submit a Pull Request?
No.

nejraselimovic, 2022-02-09:
Is this purely a documentation issue or does this order of performing API calls actually make some technical difference?

Because the JS call is left last because it is only needed for OTAA devices, while the IS, NS and AS calls are needed for both OTAA and ABP. In #772, I added OTAA and ABP tabs to differ API calls for registering these types of devices, and the easiest way to incorporate JS call part in the text without messing with the tabs is to save it for last. 

If this is purely a documentation issue, my suggestion is to close it, but if it makes a technical difference, then we can make it work somehow. 

@benolayinka fyi

benolayinka, 2022-02-14:
@htdvisser which order gets your stamp of approval?

htdvisser, 2022-02-14:
As long as the IS comes first, the order of the API calls doesn't matter that much. I have a slight preference for IS-JS-NS-AS, because (1) that's also what the CLI does and (2) because I think we should present OTAA as the default/preferred/recommended activation method, not as some extra work that people can also skip if they are too lazy.

nejraselimovic, 2022-02-16:
I rearranged the doc in #772 to meet the preferred order.

closed 2022-02-16 by nejraselimovic