TheThingsIndustries/lorawan-stack-docs #776: Document class B and C confirmed downlinks behavior

adriansmares, 2022-02-15:
#### Summary

Confirmed downlinks initiated by the network server (in other words, class B and C downlinks) have a rather peculiar behavior when confirmations are requested. We should document what the behavior is, why it is like that, and how to deal with it.

#### Why do we need this ?

Multiple users have been surprised by the behavior, since it may not match their intuition.

#### What is already there? What do you see now?

Instructions on how to schedule downlinks, confirmed and unconfirmed.

#### What is missing? What do you want to see?

I'm not entirely sure how to structure this. Maybe as a note that we attach to the class B and C sections ?

#### How do you propose to document this?

The Network Server considers a downlink as being ack(acknowledged) or nack(not acknowledged) upon receiving an uplink from the end device. The uplink has a confirmation flag attached to it (at the network layer) which confirms if the downlink has been received or not. This applies to all device classes. 
The Application Server processes this ack/nack and re-encrypts + re-enqueues the downlink if a nack occured. This works as expected in the uplink->downlink->uplink->downlink flow of class A - should a downlink not be acknowledged, the Application Server re-submits it automatically and nothing is theoretically lost.

The unexpected behavior occurs when the confirmed downlink is sent as part of a class B / class C downlink slot (basically without an uplink as a per-requisite). When such a downlink is sent, and no ack/nack is received from the end device, the Network Server **will not generate a ack/nack until an uplink arrives**. This means that if the end device sends messages at irregular intervals, the downlink will not be manually retried.

The Network Server cannot safely take the decision to re-attempt the downlink, even after a period of time, since it may loop forever if the end device is no longer in the range of the gateway / powered off. Looping a downlink indefinitely is dangerous since we consume the duty cycle of the gateway, and secondly we are just creating noise in the area for no good reason.

The application that attempts to submit this confirmed downlink must take the decision to re-enqueue the downlink on its own, and it is recommended that it does not retry the downlink indefinitely. The application can know if the Network Server has ack/nacked the downlink using the `Downlink Acked` and `Downlink Nacked` message paths (via MQTT/webhooks etc.). If such a message is received, the application doesn't have to do anything, as the Network Server will retry the downlink on its own. If no ack/nack is received by the time when the downlink should have been transmitted, it should decide if it really wants to try again or not.

#### Can you do this yourself and submit a Pull Request?

Can review.

adriansmares, 2022-02-18:
So lets look a bit at the flow for any (confirmed) downlink:

Application Server receives a confirmed downlink and processes it (payload formatters + encryption) -> Network Server receives it and processes it (MAC commands + encryption) -> Gateway Server receives it and sends it to a gateway to transmit it over the air -> End device receives it

Now since we're speaking about confirmed downlinks, the end device must respond with an `ACK`, so the flow becomes

End device generates uplink with `ACK` set -> gateway receives it and sends it to Gateway Server -> Network Server receives it and decrypts it. It generates an `downlink ack` message -> The Application Server receives this `downlink ack` message and forwards it to the integrations (MQTT, webhooks etc.)

nejraselimovic, 2022-02-18:
> You are correct, but at least there is no surprise - since there is no uplink, there can be no downlink as well. Whereas in class B/C, the expectation may be that we retry.

We have this in our documentation:

> If the device does not respond in the class-c-timeout interval (5 minutes by default), all pending downlinks will be discarded. 

Do I understand correctly, that users are confused with the `class_c/b_timeout` because they expect that this defines after which period the downlink will be resent even if the uplink with an ACK/NACK wasn't received? While it means that after that period all pending downlinks will be discarded

adriansmares, 2022-02-18:
> Do I understand correctly, that users are confused with the `class_c/b_timeout` because they expect that this defines after which period the downlink will be resent even if the uplink with an ACK/NACK wasn't received? While it means that after that period all pending downlinks will be discarded

Correct. The idea there is that discarded means that no retry will take place, and the next downlink in the queue may be transmitted.

closed 2022-02-18 by nejraselimovic