TheThingsIndustries/lorawan-stack-docs #782: Recive data from Azure IoT hub from TTI using Python or C# Application

eng-ibrahim-Rahman, 2022-02-20:
#### Summary

I am using Thing Industry with Azure IoT hub to receive the data from TTI to Aure IoT hub and I made azure integration with TTI and I am able to send a message from TTI to Azure and I receive the message successfully using Azure IOT explorer and I want to use the application to read the data that send from TTI like python application or C# application or node js I faced a problem and I didn't find any doc.

1) python return error disconnecting from MQTT when I send a message from TTI  to IoT hub device **MQTT transport disconnected with result code: 7**.
2) C# didn't detect any received message when I send a message from TTI to the IoT hub device.
3) node js didn't detect any received message when I send a message from TTI to IoT hub device.

as I need this to make an application on my machine to receive data from Azure IoT hub that send from TTI to Azure IoT hub and parse the data on my machine using the application.

NicolasMrad, 2022-02-25:
Please note that the issue you are facing is from Azure's side and not TTI. You should check with Azure on how to solve this, maybe check their documentation.

eng-ibrahim-Rahman, 2022-02-27:
Hi @NicolasMrad I checked the Azure documentation but unfortunately, I found the document of integration between Azure and TTI on the TTI portal.

NicolasMrad, 2022-03-04:
Hi @eng-ibrahim-Rahman, please note that the problem you are facing is from Azure IoT Hub's side and not TTI. You need to check Azure documentation or other resources in order to troubleshoot this. If it was from TTI side we would have helped you out with it.

eng-ibrahim-Rahman, 2022-03-09:
@NicolasMrad OK,
Thank you.

closed 2022-03-09 by NicolasMrad