TheThingsIndustries/lorawan-stack-docs #785: Document the SDKs of The Things Stack

ymgupta, 2022-02-23:
#### Summary
Add information on the SDKs available for The Things Stack.
#### Why do we need this?
To let users know about the current available SDKs of The Things Stack and the ones in the pipeline.
#### What is already there? What do you see now?
We currently have [JavaScript SDK](https://github.com/TheThingsNetwork/lorawan-stack/tree/v3.18/sdk/js) (which is used by the console) for The Things Stack.
#### What is missing? What do you want to see?
- Add information about the JS SDK and the instructions to use it.
- Information on the other languages SDKs ([Go](https://github.com/TheThingsNetwork/lorawan-stack/issues/3056), [Python](https://github.com/TheThingsNetwork/lorawan-stack/issues/3057), [C/C++](https://github.com/TheThingsNetwork/lorawan-stack/issues/3054)) are in our pipeline. Users can track the stand-alone issues for each language on GitHub.
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2022-02-24:
I'd like to provide more info on how to use the JS SDK. I don't know much about using that and the [README](https://github.com/TheThingsNetwork/lorawan-stack/tree/v3.18/sdk/js) doesn't really tell much. Can you provide more info about it or someone else from our team?

ymgupta, 2022-02-24:
CC: @kschiffer  (Tagging you based on the previous context we have [here](https://github.com/TheThingsIndustries/lorawan-stack-support/issues/100#issuecomment-666340427)).