TheThingsIndustries/lorawan-stack-docs #786: Document the Swagger file for The Things Stack APIs

ymgupta, 2022-02-23:
#### Summary
Some of The Things Stack users are facing difficulty in forming APIs for their use cases. However, with Swagger for The Things Stack APIs, users can find all the APIs with examples in one place. It would be helpful for users in the forming API.
#### Why do we need this ?
To provide a better experience for users using The Things Stack APIs.
#### What is already there? What do you see now?
[API ](https://www.thethingsindustries.com/docs/reference/api/)& [Using the API](https://www.thethingsindustries.com/docs/getting-started/api/) sections.
#### What is missing? What do you want to see?
Instructions for using the `api.swagger.json` file.
#### How do you propose to document this?
Document how users can make use of the [Swagger](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.18/api/api.swagger.json) for The Things Stack APIs in the [Using the API](https://www.thethingsindustries.com/docs/getting-started/api/) section.
#### Can you do this yourself and submit a Pull Request?
No.

johanstokking, 2022-06-24:
Blocked by https://github.com/TheThingsNetwork/lorawan-stack/issues/2297
References https://github.com/TheThingsNetwork/lorawan-stack/issues/2298
References https://github.com/TheThingsNetwork/lorawan-stack/issues/2798

htdvisser, 2022-07-20:
I've made some updates to the way we generate the Swagger file that (hopefully) fix the current issues with hex (and a few other) custom encodings.

https://github.com/TheThingsNetwork/lorawan-stack/pull/5627
https://github.com/TheThingsIndustries/lorawan-stack/pull/3327

So I don't think this is blocked anymore, and we can start investigating what's the best way to help people use our Swagger file(s).

KrishnaIyer, 2023-10-01:
I'll get back to this after https://github.com/TheThingsIndustries/product-management/issues/23

closed 2024-03-04 by KrishnaIyer