TheThingsIndustries/lorawan-stack-docs #800: Feedback on Devices documentation section.

ymgupta, 2022-03-04:
#### Summary
We had identified a few improvements in the devices documentation section.
Update the Lora Alliance hyperlinks in the [Best Practices](https://www.thethingsindustries.com/docs/devices/best-practices/) document.
Fix the hyperlink in the [Power Cycles](https://www.thethingsindustries.com/docs/devices/best-practices/#power-cycles) section in the [Best Practices](https://www.thethingsindustries.com/docs/devices/best-practices/) document.
Update the device creation CLI command in [Class C](https://www.thethingsindustries.com/docs/devices/class-c/) and [Class B](https://www.thethingsindustries.com/docs/devices/class-b/) documents.
Update the content of the `I see no Join Requests from my device in The Things Stack` scenario in the [Troubleshooting Devices](https://www.thethingsindustries.com/docs/devices/troubleshooting/) document.
#### Why do we need this?
...
#### What is already there? What do you see now?
[Best Practices](https://www.thethingsindustries.com/docs/devices/best-practices/)
[Class B](https://www.thethingsindustries.com/docs/devices/class-b/)
[Class C](https://www.thethingsindustries.com/docs/devices/class-c/)
[Troubleshooting Devices](https://www.thethingsindustries.com/docs/devices/troubleshooting/)
#### What is missing? What do you want to see?
Update the latest Lora Alliance documentation links in the below sections [here](https://www.thethingsindustries.com/docs/devices/best-practices/).
[Synchronization, Backoff, and Jitter](https://www.thethingsindustries.com/docs/devices/best-practices/#synchronization-backoff-and-jitter)
[Use a Good Random Number Generator](https://www.thethingsindustries.com/docs/devices/best-practices/#use-a-good-random-number-generator)
[Use ADR for Stationary Devices](https://www.thethingsindustries.com/docs/devices/best-practices/#use-adr-for-stationary-devices)

-------

2. Fix the hyperlink of `[Synchronization, Backoff, and Jitter](https://www.thethingsindustries.com/docs/devices/best-practices/#synchronization-backoff-jitter)` in the Power Cycles section [here](https://www.thethingsindustries.com/docs/devices/best-practices/#power-cycles).

---------

3. Update the device creation CLI command in [Class C](https://www.thethingsindustries.com/docs/devices/class-c/) and [Class B](https://www.thethingsindustries.com/docs/devices/class-b/) documents.
As we recommend OTAA devices over ABP, could you please update the existing CLI commands with CLI commands for creating OTAA devices with Class C/Clas B enabled?
FTR, device creating fails with the `no_end_device_eui (no end device EUIs set)` error when using the existing commands in the documentation. After adding the `--abp` flag, it was resolved for Class C. But, for Class B, it will still ask for `ping_slot_periodicity.value`.

-----

4. Update the content of the `I see no Join Requests from my device in The Things Stack` scenario in the [Troubleshooting Devices](https://www.thethingsindustries.com/docs/devices/troubleshooting/) document.
> If your device uses ABP, it is also possible that it is sending join requests on frequencies that The Things Stack is not listening on. See [using inappropriate frequencies](https://www.thethingsindustries.com/docs/devices/troubleshooting/#i-can-see-some-received-uplinks-in-gateway-live-data-events-but-i-do-not-see-them-in-device-events) below.

In the case of ABP devices, there will be no Joins. Please update this with the below scenario.

The possible reason could be that the device is sending join requests in non-default frequencies, and hence they are dropped at the gateway level. As per LoRaWAN Specification, the device should send the join requests in the default frequencies of the band. In this case, one has to reach out to the device manufacturer to fix the issue at the device level and make the device align with LoRaWAN Specification.

#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

ymgupta, 2022-03-09:
Hi @nejraselimovic,

Below are a couple of improvements in addition to the ones suggested earlier.

1. Update the[ ATECC608A/B Secure Elements ](https://www.thethingsindustries.com/docs/devices/atecc608a/)document.

Currently, in the ATECC608A/B Secure Elements document, the images are outdated with respect to the latest TTS console UI.

-----------

3. Move the Device creation API calls in [Using the API](https://www.thethingsindustries.com/docs/getting-started/api/#multi-step-actions) document to [Adding Devices](https://www.thethingsindustries.com/docs/devices/adding-devices/) document.

Currently, API calls for creating devices in TTS are documented in [Using the API](https://www.thethingsindustries.com/docs/getting-started/api/#multi-step-actions) document. But, it would be better if we move these API calls to the Adding[ Devices](https://www.thethingsindustries.com/docs/devices/adding-devices/) section so that customers will have all the options available to register in one place. We can add an extra `API` tab along with the console and CLI options.

ymgupta, 2022-03-14:
Hi @nejraselimovic,

Below is a troubleshooting point to add to the [Troubleshooting Devices](https://www.thethingsindustries.com/docs/devices/troubleshooting/) section.

  7. I am getting `The LoRaWAN version <MAC version> does not support the <Frequency Plan ID> frequency plan. Please choose a different MAC version or frequency plan` error while registering a device in TTS.

**Cause:** The Things Stack returns the reported error when the **frequency plan is not supported by the corresponding LoRaWAN Specification (MAC) and Regional Parameters (PHY)**.

**Resolution:**
- We suggest users to register the device in The Things Stack with the valid LoRaWAN version (MAC) and Regional Parameters version (PHY). Users can check the correct information from the devices' user manual or getting started guides. For further support, they can contact the device manufacturer.
- Users can also search the corresponding Regional Parameters document on the [LoRa Alliance](https://lora-alliance.org/search-site/) website to see if the selected frequency plan is added in that version or not.

nejraselimovic, 2022-03-18:
> Move the Device creation API calls in [Using the API](https://www.thethingsindustries.com/docs/getting-started/api/#multi-step-actions) document to [Adding [...]

@ymgupta FYI I skipped doing this for both devices and gateways, as we agreed on not doing this on our last docs sync. If we added a new tab for API in Adding devices and Adding gateways, we would then have to do it for every single feature (if not, we create a discrepancy). That would take too much time and we don't think that's needed atm. People mostly use the Console or the CLI, and on the other hand, people who use the API are usually technically skilled enough to go through the existing API documentation and examples and create a query on their own. 

> Update the[ ATECC608A/B Secure Elements ](https://www.thethingsindustries.com/docs/devices/atecc608a/)document.
>
>Currently, in the ATECC608A/B Secure Elements document, the images are outdated with respect to the latest TTS console UI.

I'm afraid I cannot complete this one, as I don't own this device. I'll make a new issue on this and try to find out who owns one, so they can update screenshots.

nejraselimovic, 2022-03-18:
Closed with #808

closed 2022-03-18 by nejraselimovic