TheThingsIndustries/lorawan-stack-docs #801: Feedback on Gateways documentation section.

ymgupta, 2022-03-05:
#### Summary
We had identified a few improvements in the Gateways documentation section.
1. Add an `API` tab in the [Adding Gateways](https://www.thethingsindustries.com/docs/gateways/adding-gateways/) section with instructions for adding gateways via the API.
2. Fix the hyperlinks in
  a. [Authentication ](https://www.thethingsindustries.com/docs/gateways/lora-basics-station/#authentication)section in the [LoRa Basics Station](https://www.thethingsindustries.com/docs/gateways/lora-basics-station/) document.
  b. [How do I use an API Key for the LBS gateways?](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#how-do-i-use-an-api-key-for-lbs-gateways) section in the [Troubleshooting Gateways](https://www.thethingsindustries.com/docs/gateways/troubleshooting/) document.

3. Reduce font size of the [Gateway shows status Disconnected, and the LED status is solid green. What do I do?](https://www.thethingsindustries.com/docs/gateways/thethingsindoorgateway/#gateway-shows-status-disconnected-and-the-led-status-is-solid-green-what-do-i-do) section in the [TTIG ](https://www.thethingsindustries.com/docs/gateways/thethingsindoorgateway/)document.

4. Update the below points in the [Troubleshooting Gateways](https://www.thethingsindustries.com/docs/gateways/troubleshooting/) guide.
  a. Add the **windows command** to generate `api.key` in the [How do I use an API Key for LBS gateways?](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#how-do-i-use-an-api-key-for-lbs-gateways) section.
  b. Update the content of the [How do I see gateway events?](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#how-do-i-see-gateway-events) section.
  c. Update the content of the [I get a “Gateway with EUI is Already Registered” error when adding a gateway](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#i-get-a-gateway-with-eui-is-already-registered-error-when-adding-a-gateway) section.
  d. Add below new troubleshooting points.
  
      i.  **What are the different packet forwarders supported in The Things Stack?**
      ii.  **I'm noticing `HTTP connect failed: NET - Failed to get an IP address for the given hostname` error in the packet forwarder logs of a LoRa Basics Station gateway. What is causing this?**
     iii. **I'm noticing `Send failed: X509 - Certificate verification failed, e.g. CRL, CA, or signature check failed` error in the packet forwarder logs of a LoRa Basics Station gateway. What is causing this?**
 
   e. Reorder the common issues and frequently asked questions.
#### Why do we need this ?
..
#### What is already there? What do you see now?
[Gateways](https://www.thethingsindustries.com/docs/gateways/)
#### What is missing? What do you want to see?

1. Add an `API` tab in the [Adding Gateways](https://www.thethingsindustries.com/docs/gateways/adding-gateways/) section with instructions for adding gateways via the API.

Currently, instructions to add gateway via APIs are documented [here](https://www.thethingsindustries.com/docs/getting-started/api/#register-a-gateway).

Since we already have Console & CLI tabs for *[Adding Gateways](https://www.thethingsindustries.com/docs/gateways/adding-gateways/)*, it would be great if we could add an API tab and move the instructions from [here](https://www.thethingsindustries.com/docs/getting-started/api/#register-a-gateway) to [Adding Gateways](https://www.thethingsindustries.com/docs/gateways/adding-gateways/).

---

2. Fix the hyperlinks in
  a. [Authentication ](https://www.thethingsindustries.com/docs/gateways/lora-basics-station/#authentication)section in the [LoRa Basics Station](https://www.thethingsindustries.com/docs/gateways/lora-basics-station/) document.
  b. [How do I use an API Key for the LBS gateways?](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#how-do-i-use-an-api-key-for-lbs-gateways) section in the [Troubleshooting Gateways](https://www.thethingsindustries.com/docs/gateways/troubleshooting/) document.

Here is the updated reference link for [Authentication Modes](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=authmodes.html) from Basic Station 2.0.6 January 2022 documentation.

---

3.  Reduce font size of the [Gateway shows status Disconnected, and the LED status is solid green. What do I do?](https://www.thethingsindustries.com/docs/gateways/thethingsindoorgateway/#gateway-shows-status-disconnected-and-the-led-status-is-solid-green-what-do-i-do) section in the [TTIG ](https://www.thethingsindustries.com/docs/gateways/thethingsindoorgateway/)document.

Update the font size of the above section.

---

4. Update the below points in the [Troubleshooting Gateways](https://www.thethingsindustries.com/docs/gateways/troubleshooting/) guide.

a. Add the **windows** command to generate api.key in the [How do I use an API Key for LBS gateways?](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#how-do-i-use-an-api-key-for-lbs-gateways) section.

Update the below **windows commands** to generate `api.key` in the [How do I use an API Key for LBS gateways?](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#how-do-i-use-an-api-key-for-lbs-gateways) section.

On Windows, you can use Command Prompt:
```
set LNS_KEY=your-lns-api-key
echo Authorization: Bearer %LNS_KEY% > lns.key
```
or PowerShell:
```
$env:LNS_KEY='your-lns-api-key'
write-output "Authorization: Bearer $env:LNS_KEY" | set-content lns.key
```

---

b. Update the content of the [How do I see gateway events?](https://www.thethingsindustries.com/docs/gateways/troubleshooting/#how-do-i-see-gateway-events) section.

> Gateway event logs can be found in the console in the gateway’s general information page. See [Working with Events](https://www.thethingsindustries.com/docs/getting-started/events/) for other ways of subscribing to events.
[... 68 lines cut ...]
- My gateway is shown as connected in the console but I don’t see any events (including the gateway connection stats). What do I do?
- My gateway is offline, but still shows as connected in the console. Why?
- My gateway is shown as disconnected after recreating with the same Gateway EUI. What do I do?
- My gateway shows an “Other cluster” status. Why?
- Why do I still see connection stats of a gateway I previously disconnected?
- When the connection on the main interface goes down, my gateway gets disconnected and it does not reconnect through the backup interface.

**Downlink Queries:**

- No downlinks are reaching my UDP gateway. Why?
- I do not see the downlink being scheduled by the Network Server. What do I do?
- When I schedule a downlink, I see the `gs.down.tx.fail` event in the Live data tab and the downlink message fails to be transmitted. Why?
- I’m noticing some errors in my UDP gateway’s packet forwarder logs. What do they represent?
- I’m noticing “radio is not emitting frame - abandoning TX, trying alternative” error in the LoRa Basics Station gateway’s packet forwarder logs. What is causing this?
- I’m seeing “no DC in band” error in the LoRa Basics Station gateway’s packet forwarder logs. What is causing this?

**Gateway Location Queries:**

- Why do my gateway’s GPS location details not show in The Things Stack Console?
- I set the gateway location manually in The Things Stack Console, why can I not see it in the gateway connection statistics?

#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No. @nejraselimovic

ymgupta, 2022-03-07:
Below are a couple of improvements in addition to the ones suggested earlier.

5. The word `endpoints` is misspelled in the [Multi-Tenancy](https://www.thethingsindustries.com/docs/reference/api/gateway_server_mqtt/#multi-tenancy) section of the [Gateway Server MQTT](https://www.thethingsindustries.com/docs/reference/api/gateway_server_mqtt/) document.

Please update it accordingly.

6. Fix the hyperlink of the **Chirpstack Gateway Bridge** in the [Packet Forwarders](https://www.thethingsindustries.com/docs/reference/packet-forwarder/) document.

Currently, the link for **Chirpstack Gateway Bridge** redirects to [TTN Packet forwarder](https://github.com/TheThingsNetwork/packet_forwarder) instead of [Chirpstack Gateway Bridge](https://www.chirpstack.io/gateway-bridge/). Update the reference link.

nejraselimovic, 2022-03-11:
> ii. I'm noticing HTTP connect failed: NET - Failed to get an IP address for the given hostname error in the packet forwarder logs of a LoRa Basics Station gateway. What is causing this?
>
> Cause: The [...]

Which URI are we talking about here?

ymgupta, 2022-03-14:
@nejraselimovic 

> Which URI are we talking about here?

It is CUPS URI.

nejraselimovic, 2022-03-18:
> @nejraselimovic
> 
> > Which URI are we talking about here?
> 
> It is CUPS URI.

Ok, but I see no CUPS info in [Port allocations](https://www.thethingsindustries.com/docs/reference/networking/#port-allocations) that you want to ref, at least not what I would expect. I was expecting to see port 443, similar as to what exists for LBS LNS. Do you happen to know why?

PR #808 (2022-03-18): doc: Update troubleshooting, spelling and link fixes
#### Summary
Closes #801 and #800.

#### Changes
Rearranging [Troubleshooting Gateways](https://www.thethingsindustries.com/docs/gateways/troubleshooting/) doc to have a more natural reading flow, spelling and link fixes, removing unnecessary bold text style, etc.

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect [... cut]
closed 2022-03-18 by nejraselimovic