TheThingsIndustries/lorawan-stack-docs #815: Improvements in the AWS ECS documentation

ymgupta, 2022-03-16:
#### Summary
We have identified a few improvements in AWS ECS documentation.
1. Update the order of the [Troubleshooting AWS ECS Deployment](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/troubleshooting/) section.
2.  Add reference for the Opsgenie service integration with the AWS CloudWatch in [Opsgenie Alarms](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/deployment/#opsgenie-alarms-optional) section.
3. Add rate-limiting cloud formation template in the [Deployment](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/deployment/) section.
#### Why do we need this ?
...
#### What is already there? What do you see now?
[AWS ECS](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/) documentation.
#### What is missing? What do you want to see?
1. Update the order of the [Troubleshooting AWS ECS Deployment](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/troubleshooting/) section.

Currently, the Troubleshoot section is in the middle of the other sections. Please move it to the end, i.e., after the [Monitoring](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/monitoring/) section.

---------------------------------------------------------------------------
2. Add reference for the Opsgenie service integration with the AWS CloudWatch in [Opsgenie Alarms](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/deployment/#opsgenie-alarms-optional) section.

Add the following reference to show users how to integrate the Opsgenie service with AWS CloudWatch and get Opsgenie CloudWatch API Key and Opsgenie Alerts Endpoint.
Ref: https://support.atlassian.com/opsgenie/docs/integrate-opsgenie-with-amazon-cloudwatch/

----------------------------------------------------------------
3. Add rate-limiting cloud formation template in the [Deployment](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/deployment/) section.

Currently, there is no rate-limiting cloud formation template in the deployment section. Please add the rate-limiting template section
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

michalborkowski96, 2022-03-18:
Docs should update within an hour.

ymgupta, 2022-03-21:
Additional Scenario:

-----------
Database migrations when updating a deployment.

**Scenario:** A user is updating a deployment from an older version that is 2 patches & 1 minor version behind the latest version. As per release notes, it is mentioned to do database migration of IS in 2 versions. In this case, the user is confused if they need to perform database migrations twice (on each version) or only once (after upgrading to the target version).

**Resolution:** One-time execution (after upgrading to the target version) is enough for all the schema migrations in between the upgraded versions (`v3.a.b` to `v3.x.y`). I.e., the user can upgrade a CloudFormation template to `v3.x.y` directly and run the database migration once. It will do the necessary database migrations required from `v3.a.b` to `v3.x.y`.

---

@michalborkowski96, please add this as a `Note` point in the Updating documentation.

michalborkowski96, 2022-03-29:
A new `note` section should appear at https://www.thethingsindustries.com/docs/getting-started/aws/ecs/updating/ within an hour

closed 2022-03-30 by ymgupta