TheThingsIndustries/lorawan-stack-docs #817: Improvements in AWS Marketplace AMI documentation

ymgupta, 2022-03-16:
#### Summary
We have identified a few improvements in AWS Marketplace AMI documentation.
1. Update the order of the [Troublshooting AWS AMI Deployment](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/) section.
2. Add a hyperlink to the Gateway issues troubleshooting in the [My Gateway doesn’t connect. What do I do?](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/#my-gateway-doesnt-connect-what-do-i-do) section.
3. Add a hyperlink to the Device issues troubleshooting in the [My device doesn’t join. How do I fix this?](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/#my-device-doesnt-join-how-do-i-fix-this) section.
4. Fix the hyperlink for the support page in [Professional Support](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/#professional-support) section.
#### Why do we need this?
....
#### What is already there? What do you see now?
[AWS Marketplace AMI](https://www.thethingsindustries.com/docs/getting-started/aws/ami/)
#### What is missing? What do you want to see?
1. Update the order of the [Troublshooting AWS AMI Deployment](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/) section.

Currently, the Troubleshoot section is in the middle of the other sections. Please move it to the end, i.e., after the [Application Server Telemetry](https://www.thethingsindustries.com/docs/getting-started/aws/ami/application-server-telemetry/) section.

-------------------------------------------------
2. Add a hyperlink to the Gateway issues troubleshooting in the [My Gateway doesn’t connect. What do I do?](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/#my-gateway-doesnt-connect-what-do-i-do).

Please redirect users to the [Troubleshooting Gateways](https://www.thethingsindustries.com/docs/gateways/troubleshooting/) document from this section.

-------------------------------------------------
3. Add a hyperlink to the Device issues troubleshooting in the [My Gateway doesn’t connect. What do I do?](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/#my-device-doesnt-join-how-do-i-fix-this).

Please redirect users to the [Troubleshooting Devices](https://www.thethingsindustries.com/docs/devices/troubleshooting/#my-device-will-not-join-what-do-i-do) document from this section.

-----------------------------------------------------
4. Fix the hyperlink for the support page in the [Professional Support](https://www.thethingsindustries.com/docs/getting-started/aws/ami/troubleshooting/#professional-support).

The support page hyperlink returns the `Page Not Found` error. Update the hyperlink with the correct support page URL.
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

ymgupta, 2022-03-19:
In addition to the changes suggested,

5. Remove the [Multi-tenancy](https://www.thethingsindustries.com/docs/getting-started/aws/ami/best-practices/#multi-tenancy) section in the [AWS AMI Best practices](https://www.thethingsindustries.com/docs/getting-started/aws/ami/best-practices/) section.

As per the [Deployment options](https://www.thethingsindustries.com/deployment/), the AWS AMI deployment doesn't support multi-tenancy. So, please remove the Multi-tenancy section.

nejraselimovic, 2022-03-23:
> Currently, the Troubleshoot section is in the middle of the other sections. Please move it to the end, i.e., after the [Application Server Telemetry](https://www.thethingsindustries.com/docs/getting-started/aws/ami/application-server-telemetry/) section.

I don't see this.
[image]

PR #826 (2022-03-23): doc: Improve AWS AMI troubleshooting doc
#### Summary
Closes #817 

#### Changes
Addressed a few minor improvements (links).

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x]  [... cut]
closed 2022-03-23 by nejraselimovic