TheThingsIndustries/lorawan-stack-docs #819: Improvements in Billing With Stripe documentation

ymgupta, 2022-03-17:
#### Summary
Update the Billing with Stripe documentation according to the latest UI changes in the Stripe Dashboard.
#### Why do we need this?
To update the documentation with the latest UI changes in the stripe.
#### What is already there? What do you see now?
[Billing With Stripe](https://www.thethingsindustries.com/docs/reference/stripe/)
#### What is missing? What do you want to see?
The UI and naming for a few options in the Stripe dashboard seem updated recently (e.g. **+New** changed as **+Add Product** in the Products Page). Please update the [Products and Pricing Plans](https://www.thethingsindustries.com/docs/reference/stripe/products/) and [Customers](https://www.thethingsindustries.com/docs/reference/stripe/customers/) sections in TTS documentation according to the latest changes in Stripe dashboard UI.

cc: @adriansmares 
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2022-03-18:
I'm afraid that I don't have access to Stripe, nor do I know how billing system works. @adriansmares can you pick this up?

ymgupta, 2022-04-22:
FTR, please also document the insights mentioned [here](https://github.com/TheThingsIndustries/lorawan-stack-support/issues/757#issuecomment-1081896831).