TheThingsIndustries/lorawan-stack-docs #820: Improvements in the Rate Limiting documentation

ymgupta, 2022-03-17:
#### Summary
We have identified a few improvements in Rate Limiting documentation.
Update the misconfigured links in the  [Rate Limited Entities](https://www.thethingsindustries.com/docs/reference/rate-limiting/#rate-limited-entities) section
Update the text formatting in [OAuth Server, Account App, Gateway Configuration Server, Interop Server, Metrics, Health, pprof](https://www.thethingsindustries.com/docs/reference/rate-limiting/#oauth-server-account-app-gateway-configuration-server-interop-server-metrics-health-pprof) section
#### Why do we need this?
...
#### What is already there? What do you see now?
[Rate Limiting](https://www.thethingsindustries.com/docs/reference/rate-limiting/) documentation
#### What is missing? What do you want to see?
1. Clicking on the text content of a section is redirecting to the next section. Please check and fix this.

**Example:**

Hover and click the space between the `Key` and `Example` attributes of the `Application Server MQTT Client Connections` subsection.

https://user-images.githubusercontent.com/9304207/158823419-341a6498-8d2a-4ed5-ab6a-0dac5963aea9.mp4

On clicking, it is redirecting to the next subsection, `Application Server MQTT Downlink Traffic`.

Please ignore if this is as intended.

------------------

2. Update the text formatting in the **Description** of [OAuth Server, Account App, Gateway Configuration Server, Interop Server, Metrics, Health, pprof](https://www.thethingsindustries.com/docs/reference/rate-limiting/#oauth-server-account-app-gateway-configuration-server-interop-server-metrics-health-pprof) subsection
We see that the text formatting is displayed as below. Fix it if possible.
```
`{{ Server }}` is `oauth`, `account`, `gcs`, `interop`, `metrics`, `health`, `pprof`
```
#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2022-03-23:
These are all rendering issues so I'm assigning this to @benolayinka

KrishnaIyer, 2023-12-19:
This content is out of date and those fixes are applied now.

closed 2023-12-19 by KrishnaIyer