TheThingsIndustries/lorawan-stack-docs #824: Document the missing stack configuration options of TTS

ymgupta, 2022-03-22:
#### Summary
The Things Stack binary is configurable with many different options. TTS Enterprise and TTS Cloud deployments offer some additional configuration options compared to Open Source deployments. These additional options are not part of the current document.
Ref: https://www.thethingsindustries.com/docs/reference/configuration/
#### Why do we need this?
The documentation must be up-to-date with all the available configuration options for TTS enterprise and cloud deployments. So, users can know the available options.
#### What is already there? What do you see now?
[Configurations](https://www.thethingsindustries.com/docs/reference/configuration/) for each component in the TTS.
#### What is missing? What do you want to see?
All the available configuration options of TTS, including the options specific to TTS Enterprise and Cloud deployments, along with their description.
#### How do you propose to document this?
- Document the missing configuration options of TTS Enterprise & Cloud deployments and their description. Use the command `tti-lw-stack config --yml` to get all the available configuration options.
- It would be good to document the newly added configuration options (if any) whenever there is a new release.
#### Can you do this yourself and submit a Pull Request?
No

nejraselimovic, 2022-03-23:
Enterprise and Cloud configuration options are actually included in docs and labeled with a deployment tag to note that it's for those deployments only. For example: https://www.thethingsindustries.com/docs/reference/configuration/tenant-billing-server/. However, some options could easily be missing.

I went through all configuration options maybe a month ago and added all that were missing. I'm going to check again, but this is really inefficient way of tracking them. If you have certain config options that you already saw are missing, please list them here.

> It would be good to document the newly added configuration options (if any) whenever there is a new release.

I completely agree with this, so I'm going to raise an issue to require documenting configuration options whenever new ones are added. This should be a responsibility of a developer who adds it, because often I have no idea what those options mean.

ymgupta, 2022-03-23:
> I completely agree with this, so I'm going to raise an issue to require documenting configuration options whenever new ones are added. This should be a responsibility of a developer who adds it, because often I have no idea what those options mean.

Sure. Also, it would be good to automate the process similar to the one that we are following for documenting CLI/API options after each release.

Below are some options that are missing in the documentation.

**Application Server Options:**

```
--as.aws.iot.telemetry                                                          Enable publishing telemetry to AWS IoT
--as.aws.region string                                                           AWS region (optional)
--as.distribution.timeout duration                                              Wait timeout of an empty subscription set (default 1m0s)
--as.end-device-metadata-storage.location.timeout duration                      Timeout of the end device retrival operation (default 5s)
--as.fetcher.circuit-breaker.timeout duration                                   Timeout after which the circuit breaker closes
--as.fetcher.timeout duration                                                   Timeout of the end device retrival operation
--as.link-mode string                                                           Deprecated - mode to link applications to their Network Server (all, explicit)
--as.packages.azure-iot-hub.cluster.application-server string                   Application Server cluster address (default "localhost")
--as.packages.azure-iot-hub.cluster.join-server string                          Join Server cluster address (default "localhost")
--as.packages.azure-iot-hub.cluster.network-server string                       Network Server cluster address (default "localhost")
--as.packages.timeout duration                                                  Message processing timeout (default 10s)
--as.pubsub.providers strings                                                   Controls the status of each provider (enabled, disabled, warning) (default [mqtt=enabled,nats=enabled])
```

**Identity Server Options:**
```
--is.database-max-idle-conns int                                                Maximum number of idle database connections (default 10)
--is.database-max-open-conns int                                                Maximum number of open database connections (default 20)
--is.user-login.disable-credentials-login                                       Disable login with credentials
```

**Crypto Server Options:**

```
--cs.microchip.aws.parts-table string                                           DynamoDB table of parts
--cs.microchip.aws.region string                                                Region
--cs.microchip.enable                                                           
--cs.microchip.expose-root-keys                                                 Expose root keys
--cs.microchip.gcp.part-kind string                                             Datastore part kind
--cs.microchip.gcp.project-id string                                            Project ID
--cs.microchip.reload-keys-interval duration                                    Interval to reload keys
--cs.microchip.source string                                                    Source keys (aws, gcp)
```

nejraselimovic, 2022-03-25:
I can't see any crypto server options (`cs.microchip.....`) when I run `tti-lw-stack config`. Can you tell me where to find this?