TheThingsIndustries/lorawan-stack-docs #834: Document connecting Cisco IXM Gateway using Basics Station

ymgupta, 2022-03-26:
#### Summary
Cisco supports LoRa Basics Station with the latest firmware. The current documentation doesn't have information on connecting Gateway using LBS with TTS. Ref: https://www.thethingsindustries.com/docs/gateways/ciscowirelessgateway/
#### Why do we need this ?
To connect Gateway with TTS using LBS.
#### What is already there? What do you see now?
Guide on how to connect the Cisco Wireless Gateway to The Things Stack using UDP. Ref: https://www.thethingsindustries.com/docs/gateways/ciscowirelessgateway/
#### What is missing? What do you want to see?

- To connect Cisco Gateway in UDP protocol, **Cisco gateways appear to use a proprietary format, with additional fields for their FPGA**. The instructions on https://www.thethingsindustries.com/docs/gateways/ciscowirelessgateway/ won’t work with our `global_conf.json` file. We need to update the documentation [here](https://www.thethingsindustries.com/docs/gateways/ciscowirelessgateway/#retrieve-configuration-from-the-gateway-configuration-server).

- Add a guide to connecting the Cisco Gateway with TTS using LBS.

#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No. @benolayinka

nejraselimovic, 2022-06-23:
@ymgupta do you have any useful documentation we can use here? I can't seem to find any guide for using this gateway with LBS

ymgupta, 2022-07-01:
Hi @nejraselimovic,

There is [documentation](https://www.cisco.com/c/en/us/td/docs/routers/interface-module-lorawan/software/configuration/guide/b_lora_scg/lrr.html#Cisco_Task.dita_a86d318a-6ea7-46d4-b764-b776ad1912ad) from Cisco on configuring the packet forwarder, but it is generic.

Also, The Things Stack supports only [TLS server authentication and client token](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=authmodes.html) mode at the moment. Cisco added support for this Authentication Mode from the `v2.3.0` release. So, the **gateway should be running with minimum firmware version** [`2.3.0`](https://www.cisco.com/c/en/us/td/docs/routers/interface-module-lorawan/software/release/notes/b_lorawan_rn_2_3_0.html).

nejraselimovic, 2022-07-20:
I looked into this but I'm afraid I can't do anything here without using the device.

KrishnaIyer, 2023-10-01:
This is being bounced around a bit. I think @adriansmares or myself can pick it up and run tests with our test gateway.

PR #1288 (2024-04-03): doc: Modify Cisco IXM doc
#### Summary
Closes #834 

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Style Guidelines: Documentation obeys style guidelines in [CONTRIBUTING]( [... cut]
closed 2024-04-03 by nejraselimovic