TheThingsIndustries/lorawan-stack-docs #856: Add additional setup step to the 'Connect Tektelic Kona Micro with Lora Basics™ Station' guide

gregthecloudandcompass, 2022-04-20:
**Please note:**
I'm fairly new to TTN and LoRa and this might just be user error on my part. The fix worked but my solution might be wrong.
I would need someone more knowledgeable than me to know if the below fix makes sense.

#### Summary

There is a setup step only mentioned when using KonaFT Tool guide but I think it is required for setting up Lora Basics Station also.

#### Why do we need this ?

To improve the guide and make the setup more seamless.

#### What is already there? What do you see now?

Step 4. under the 'Using KonaFT Tool' heading 
https://www.thethingsindustries.com/docs/gateways/tektelickonamicro/#using-konaft-tool 

Page linked from step 4.
https://www.thethingsindustries.com/docs/gateways/tektelickonamicro/udp/

Page to be modified
https://www.thethingsindustries.com/docs/gateways/tektelickonamicro/lbs/

#### What is missing? What do you want to see?

I believe the Lora Basics station setup page (https://www.thethingsindustries.com/docs/gateways/tektelickonamicro/lbs/) should also tell the user to complete the steps on the page below
https://www.thethingsindustries.com/docs/gateways/tektelickonamicro/udp/

#### How do you propose to document this?

Add a section on the Lora Basics Station setup page to tell users to follow the steps on the (https://www.thethingsindustries.com/docs/gateways/tektelickonamicro/udp/) page.

#### Can you do this yourself and submit a Pull Request?

I'm not sure

nejraselimovic, 2023-10-13:
@KrishnaIyer please correct me if I'm wrong, but I don't think it makes sense to connect the gateway using UDP in order to connect it using LBS. I think we can close this issue

KrishnaIyer, 2023-10-16:
Yes that's indeed correct. @gregthecloudandcompass: LBS and UDP are different gateway forwarder protocols and are mutually exclusive.

closed 2023-10-16 by KrishnaIyer