TheThingsIndustries/lorawan-stack-docs #921: Update ADR documentation based on the latest changes in TTS

ymgupta, 2022-07-16:
#### Summary
Update the ADR documentation with the recent changes made to the ADR configuration in TTS. Please refer to the below issues for context.
Ref: https://github.com/TheThingsNetwork/lorawan-stack/pull/5353 & https://github.com/TheThingsNetwork/lorawan-stack/pull/5326

#### Why do we need this ?
To let users know how to configure ADR for their devices in TTS.

#### What is already there? What do you see now?
[Adaptive Data Rate](https://www.thethingsindustries.com/docs/reference/adr/)

#### What is missing? What do you want to see?
Instructions on how to configure ADR and different options that are available in TTS.

#### How do you propose to document this?
Document the latest configuration options added in TTS regarding ADR and their usage instructions.

CC: #Adrian

#### Can you do this yourself and submit a Pull Request?
No

NicolasMrad, 2022-07-18:
@nejraselimovic do you have the know-how to do this? Or should Adrian help out?

ymgupta, 2023-06-21:
Hi @nejraselimovic,

The Things Stack [v3.19.0](https://www.thethingsindustries.com/docs/whats-new/3.19.0/) introduced an enhanced ADR (Adaptive Data Rate) settings API, providing greater control over the ADR algorithm. Please update the new ADR settings in the [MAC Settings](https://www.thethingsindustries.com/docs/devices/mac-settings/#enable-adr) and [ADR](https://www.thethingsindustries.com/docs/reference/adr/#how-adr-works) documentation.

**For example:**

   - In the [Mac Settings ](https://www.thethingsindustries.com/docs/devices/mac-settings/#enable-adr)documentation, we see the below command to enable the ADR usage, and this command was deprecated in The Things Stack v3.19.0. 
```
ttn-lw-cli end-devices set <app-id> <device-id> --mac-settings.use-adr=true 
```
- Below is the example CLI command with one of the new ADR setting, i.e., dynamic mode.

```
ttn-lw-cli end-devices set <app-id> <device-id> --mac-settings.adr.mode.dynamic
```

**FYI:**

We see that there are couple of queries from The Things Stack users regarding the new ADR settings. We are redirecting them to [The Things Stack v3.19.0 release notes](https://www.thethingsindustries.com/docs/whats-new/3.19.0/), for detailed information on new ADR settings. Could you please prioritize this documentation?

cc: @adriansmares

nejraselimovic, 2023-07-09:
Hi @ymgupta, I discussed this request few weeks ago with our product manager but he left the company in the meantime so we kinda lost track. @adriansmares I think I will need your help with this one. Can you provide some overview of the content that needs to be covered and some basic info? Or if there's too much to be written, we can have a sync so I can start working on this asap

nejraselimovic, 2023-07-09:
Hi @ymgupta, I'll start working on this right now, but @adriansmares I think I will need your help. Can you pls provide some general overview of the content that needs to be covered? We can also have a sync if needed

adriansmares, 2023-07-17:
Hey @nejraselimovic,

Fundamentally the ADR algorithm did not change, but we've done large efforts to try to make the algorithm more configurable and also did some fixes along the way.

The historical tuning knobs that are also documented here were `use_adr` and `adr_margin`. In the scenic route, `use_adr` would control on whether or not the ADR algorithm was enabled, and `adr_margin` would decide the margin that the algorithm uses while increasing the data rate.

The problem with the `use_adr` knob is that setting `use_adr` to false does not actually disable ADR, but rather it enables the 'custom ADR' behavior in which an external service is expected to edit the data rate index, transmission power index and number of transmissions - there was no way to actually disable ADR properly.

As such, `use_adr` and `adr_margin` have been deprecated in favor of the ADR 'modes':

```  
      --mac-settings.adr.mode.disabled                                                        
      --mac-settings.adr.mode.dynamic                                                         
      --mac-settings.adr.mode.dynamic.channel-steering.mode.disabled                          
      --mac-settings.adr.mode.dynamic.channel-steering.mode.lora-narrow                       
      --mac-settings.adr.mode.dynamic.margin float32                                          
      --mac-settings.adr.mode.dynamic.max-data-rate-index string                              allowed values: 0, DATA_RATE_0, 1, DATA_RATE_1, 2, DATA_RATE_2, 3, DATA_RATE_3, 4, DATA_RATE_4, 5, DATA_RATE_5, 6, DATA_RATE_6, 7, DATA_RATE_7, 8, DATA_RATE_8, 9, DATA_RATE_9, 10, DATA_RATE_10, 11, DATA_RATE_11, 12, DATA_RATE_12, 13, DATA_RATE_13, 14, DATA_RATE_14, 15, DATA_RATE_15
      --mac-settings.adr.mode.dynamic.max-nb-trans uint32                                     
      --mac-settings.adr.mode.dynamic.max-tx-power-index uint32                               
      --mac-settings.adr.mode.dynamic.min-data-rate-index string                              allowed values: 0, DATA_RATE_0, 1, DATA_RATE_1, 2, DATA_RATE_2, 3, DATA_RATE_3, 4, DATA_RATE_4, 5, DATA_RATE_5, 6, DATA_RATE_6, 7, DATA_RATE_7, 8, DATA_RATE_8, 9, DATA_RATE_9, 10, DATA_RATE_10, 11, DATA_RATE_11, 12, DATA_RATE_12, 13, DATA_RATE_13, 14, DATA_RATE_14, 15, DATA_RATE_15
      --mac-settings.adr.mode.dynamic.min-nb-trans uint32                                     
      --mac-settings.adr.mode.dynamic.min-tx-power-index uint32                               
      --mac-settings.adr.mode.static
```

The disabled mode does what it says - it completely disables ADR and allows the end device to choose on its own the data rate used and the transmission power index (what `use_adr=false` should have been).

The static mode is meant to replace the old 'custom ADR' mode. The main difference here is that the settings done here will persist even after the end device joins the network again.

The dynamic mode is meant to be equivalent the normal ADR mode (what `use_adr=true` was and still is):
- The `margin` parameter is equivalent to `adr_margin`.
- The channel steering modes relate to behavior that occurs in the US915 and AU915 bands. In these bands the end device may join via what we call a 'LoRa wide channel' (a 500kHz bandwidth channel). By default the LNS will keep the end device in this singular channel (i.e. disabled mode), but when the `lora-narrow` mode is used, the end device will be steered towards the 8 125kHz channels.
- The `min` and `max` settings allow users to clamp (both floor and ceil) certain parameters, such as the data rate index, transmission power index and number of transmissions.

An important closing remark is that the two ways of configuring ADR are mutually exclusive. In order to use the `mac-settings.adr.mode` API, the fields `use_adr` and `adr_margin` need to be unset first. This can be done using the `--unset use-adr,adr-margin` flag.

PR #1146 (2023-08-06): doc: Update ADR documentation
#### Summary
Closes #921 

#### Checklist
- [ ] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [ ] Style Guidelines: Documentation obeys style guidelines in [CONTRIBUTI [... cut]
closed 2023-08-06 by nejraselimovic