TheThingsIndustries/lorawan-stack-docs #946: Minor changes required in The Things Stack configuration files

ymgupta, 2022-09-03:
#### Summary

We have identified a few changes in The Things Stack configuration files `docker-compose.yml` and `ttn-lw-stack-docker.yml`

1. Update the NOC store URI in `docker-compose.yml`
2. Update the NOC component configuration in `ttn-lw-stack-docker.yml`

#### Why do we need this ?

...

#### What is already there? What do you see now?

NOC component configuration in [docker-compose.yml](https://www.thethingsindustries.com/docs/getting-started/installation/configuration/docker-compose-enterprise.yml) and [ttn-lw-stack-docker.yml](https://www.thethingsindustries.com/docs/getting-started/installation/configuration/ttn-lw-stack-docker-enterprise.yml).

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

1. Update the NOC store URI in `docker-compose.yml`.

In the [docker-compose.yml](https://www.thethingsindustries.com/docs/getting-started/installation/configuration/docker-compose-enterprise.yml) file, the NOC store database URI is configured with `timescaledb` instead of `postgres`. Please update the `TTN_LW_NOC_STORE_DATABASE_URI ` configuration as **expected configuration** in the `docker-compose.yml` file.

**Current Configuration:**

```
TTN_LW_NOC_STORE_DATABASE_URI: postgres://root:root@timescaledb:5432/ttn_lorawan?sslmode=disable
```

**Expected configuration:**

```
TTN_LW_NOC_STORE_DATABASE_URI: postgres://root:root@postgres:5432/ttn_lorawan?sslmode=disable
```

2. Update the NOC component configuration in `ttn-lw-stack-docker.yml`.

The `noc.store.database-uri` in the [ttn-lw-stack-docker.yml](https://www.thethingsindustries.com/docs/getting-started/installation/configuration/ttn-lw-stack-docker-enterprise.yml) as **current configuration** mentioned below. So, please update as per the **expected configuration** in the `ttn-lw-stack-docker.yml` file.

**Current Configuration**

```
store:
   database-uri: "postgres://root:root@postgres:5432/ttn_lorawan_dev?sslmode=disable"
```
**Expected Configuration**
```
store:
   database-uri: "postgres://root:root@postgres:5432/ttn_lorawan?sslmode=disable"
```

3. The `noc.ui.canonical-url` is misconfigured in the [ttn-lw-stack-docker.yml](https://www.thethingsindustries.com/docs/getting-started/installation/configuration/ttn-lw-stack-docker-enterprise.yml) file. Please update as expected configuration in the configuration file.

**Current configuration:**
```
noc:
  ui:
   canonical-url: "https://thethings.example.com/console"
```

**Expected configuration:**
```
noc:
  ui:
   canonical-url: "https://thethings.example.com/noc"
```

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

ymgupta, 2022-10-26:
Hi @nejraselimovic,

The Things Stack Enterprise users are reporting issues with NOC initialization. The issue is due to a mismatch of the NOC database URI configuration in the docker-compose.yml file. Could you please prioritize this documentation request?

ymgupta, 2022-10-31:
In addition to the above changes, please add the below configuration.

As per the new end device registration flow in `3.22.0`, The Things Stack Console should require the endpoint for the device claiming server. Please add the following [lines](https://github.com/TheThingsNetwork/lorawan-stack/blob/07b1ffb1a54a56532aea6ebf4adfc2d38c636fca/config/stack/ttn-lw-stack-docker.yml#L95-L96) to the configuration in the `console.ui` section of the `ttn-lw-stack-docker.yml` file.
```
dcs:
    base-url: 'https://thethings.example.com/api/v3'
```

Ref: https://github.com/TheThingsIndustries/lorawan-stack-support/issues/873#issuecomment-1291801820

NicolasMrad, 2022-11-29:
https://github.com/TheThingsNetwork/lorawan-stack/issues/5948

NicolasMrad, 2022-11-29:
@nejraselimovic can you please put this on high priority

nejraselimovic, 2022-11-30:
> @nejraselimovic can you please put this on high priority

Sure, will handle this today

closed 2022-12-12 by johanstokking