TheThingsIndustries/lorawan-stack-docs #956: Missing contents in code block in installation docs

tkraner, 2022-09-15:
#### Summary

File "doc/content/getting-started/installation/certificates.md" is referencing Code from the wrong lines of "doc/content/getting-started/installation/configuration/docker-compose-custom-certificates.yml", resulting in an empty code block in the docs.

#### Why do we need this ?

Re-adding info that was lost due to a commit that affected the file at "doc/content/getting-started/installation/configuration/docker-compose-custom-certificates.yml".

#### What is already there? What do you see now?

Empty code block, observable at [https://www.thethingsindustries.com/docs/getting-started/installation/certificates/#using-custom-certificates](https://www.thethingsindustries.com/docs/getting-started/installation/certificates/#using-custom-certificates)

#### What is missing? What do you want to see?

The Certificate Docs (doc/content/getting-started/installation/certificates.md) seem to be missing contents in a code block right at the beginning of the "Using Custom Certificates"-section on lines 42 to 44

#### How do you propose to document this?

Current:
```
{{< highlight yaml "linenos=table,linenostart=82" >}}
{{< readfile path="/content/getting-started/installation/configuration/docker-compose-custom-certificates.yml" from=82 to=97 >}}
{{< /highlight >}}
```
Where it should probably be referencing the following:
```
{{< highlight yaml "linenos=table,linenostart=66" >}}
{{< readfile path="/content/getting-started/installation/configuration/docker-compose-custom-certificates.yml" from=66 to=79 >}}
{{< /highlight >}}
```

#### Can you do this yourself and submit a Pull Request?

Yes.

nejraselimovic, 2022-09-18:
@tkraner I can also take this along other docs improvements. If you want me to do it, please assign me

tkraner, 2022-09-19:
> @tkraner I can also take this along other docs improvements. If you want me to do it, please assign me

@nejraselimovic That'd be great thanks, however I don't have the necessary permissions to assign you.

PR #959 (2022-09-27): doc: Fix configuration references in code blocks
#### Summary
Closes #956 

#### Changes
Fixing references to sections from `ttn-lw-stack.yml` and `docker-compose.yml` that were slightly messed up by changing these files.

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the ` [... cut]
closed 2022-09-27 by nejraselimovic