TheThingsIndustries/lorawan-stack-docs #964: Icons being rendered in front of the main page

NicolasMrad, 2022-10-04:
#### Summary
As per Ivan: A weird behaviour - each section I click on and then return to the main page renders an icon that I pressed to go to the category in front of the main page.
If I clicked the name of the category to go to the page - an empty image is rendered.
Did not check if this happens elsewhere.

https://drive.google.com/file/d/17mfYn26661KVjg01oMG-SnVA0MI7sBS6/view?usp=sharing

#### Why do we need this ?

...

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

...

nejraselimovic, 2022-10-09:
I cannot reproduce this, it works as intended on my Firefox - when I click back, it takes me back to the docs homepage. @NicolasMrad can you try? It might be related to caching or browser type.

NicolasMrad, 2022-10-10:
I tried this on Chrome and Safari and below are the results:

- Chrome: The first time I press on a category the icon animation shows normally and smoothly, press back and press the same category again the icon appears but with a very fast animation and disappears.
- Safari: The flow is reversed on Safari, press on a category and there's no icon appearing, press back and the icon appears with it's animation. Press again on the same category and press back, the icon appears full screen and gets stuck. The only way to move forward is to click anywhere on the screen to remove the icon.

[image]

nejraselimovic, 2022-10-16:
> I tried this on Chrome and Safari and below are the results:
>
>     * Chrome: The first time I press on a category the icon animation shows normally and smoothly, press back and press the same [...]

I see. @kschiffer you said that this is because we use the `figure` tag to display these illustrations. Can you tell me what could be the solution here?

NicolasMrad, 2023-04-28:
@kschiffer can you take a look at this and check out what the problem is, because apparently it's a front-end issue

KrishnaIyer, 2023-05-01:
I think this issue is stale. I have removed all these icons in the restructuring. Please check if this issue still occurs.

nejraselimovic, 2023-05-07:
Krishna is right, we no longer have these icons so I'm going to close this

closed 2023-05-07 by nejraselimovic