TheThingsIndustries/lorawan-stack-docs #967: Incorrect link is added to the Distribution Type icons

pradeeka7, 2022-10-09:
#### Summary

An incorrect link is added to the **Distribution Type** icons.

#### Why do we need this ?

The link is incorrect and the page does not exist.

#### What is already there? What do you see now?

Here is an example:
[image]
https://www.thethingsindustries.com/docs/reference/components/network-operations-center/

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

The correct link should be https://accounts.thethingsindustries.com/fee-calculator

#### Can you do this yourself and submit a Pull Request?

...

nejraselimovic, 2022-10-22:
There is a PR for this https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/976 waiting for approvals, I wasn't sure if we really want to point to the fee calculator from those icons

nejraselimovic, 2022-10-30:
Closing this since #976 was merged.

closed 2022-10-30 by nejraselimovic