TheThingsIndustries/lorawan-stack-docs #986: Add `how to check billing cycle/billing frequency and the next renewal date` FAQ to the Subscription and Billing Management docs

ymgupta, 2022-11-05:
#### Summary

To track the next renewal date of TTI Subscriptions, document the methods for checking the billing cycle/billing frequency.

#### Why do we need this ?

Assist users to know the next renewal date of their TTI subscription billing.

#### What is already there? What do you see now?

[Subscription and Billing Management](https://www.thethingsindustries.com/docs/getting-started/billing-and-subscription-management/).

#### What is missing? What do you want to see?

Describe how to track the next renewal date based on the billing cycle/frequency in the Subscription and Billing [FAQs](https://www.thethingsindustries.com/docs/getting-started/billing-and-subscription-management/#faq) section.

#### How do you propose to document this?

**Insights:**

Generally, users can check the billing frequency at the bottom of the subscription overview page.

For example, refer to the below screenshot from the documentation that shows the next renewal date on `16th November 2022`.

[image]

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

nejraselimovic, 2022-11-15:
Hmm, this is not really intuitive, the line says "Billed to Visa Card <card-number> on the 16th November 2022" and this was actually billed on 16th October. @felixheins should this line contain the date that the invoice was billed on or the next renewal? Because if it should point to the renewal date, it's not really clear. 

Also, we have some kind of emails to remind users on the renewal date or something, right?

felixheins, 2022-11-16:
This probably shows the date for the next invoice payment. So it would move up every month. 
@mjamescompton, is that correct?

KrishnaIyer, 2023-08-10:
We need to check SM documentation as a whole. I'll close this and create an issue for better SM docs.

closed 2023-08-10 by KrishnaIyer