TheThingsIndustries/lorawan-stack-docs #988: Document connecting a generic node in TTS

ymgupta, 2022-11-05:
#### Summary
Document how to configure and connect generic node in The Things Stack.

#### Why do we need this ?
To assist users in configuring and connecting the generic node to The Things Stack.

#### What is already there? What do you see now?
- [The Things Industries documentation on devices](https://www.thethingsindustries.com/docs/devices/)
- [Generic Node documentation](https://www.genericnode.com/docs/)
#### What is missing? What do you want to see?
A guide to configure and connect the generic node to TTS in the device guide. Ref: https://www.thethingsindustries.com/docs/devices/

#### How do you propose to document this?
...

#### Can you do this yourself and submit a Pull Request?
No, @nejraselimovic

pradeeka7, 2022-11-15:
> @NicolasMrad can you assign someone who owns a generic node to document this?

@NicolasMrad I can work on this.

nejraselimovic, 2022-11-16:
> > @NicolasMrad can you assign someone who owns a generic node to document this?
> 
> @NicolasMrad I can work on this.

That would be great @pradeeka7, thanks!

pradeeka7, 2023-02-10:
I have submitted a PR for this: https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1037

pradeeka7, 2023-05-02:
@NicolasMrad Can you close this issue?

closed 2023-05-02 by KrishnaIyer