TheThingsIndustries/lorawan-stack-docs #998: Update Kerlink docs

benolayinka, 2022-11-24:
#### Summary

Kerlink docs apparently use an outdated script. Let's remove the script and just use generic kerlink instructions.

#### Can you do this yourself and submit a Pull Request?

@nejraselimovic

KrishnaIyer, 2023-10-01:
I'll pick this up myself. I can use one of our test gateways.

pospospos2007, 2024-02-23:
Does Kerlink support base station? do you guys have any idea on this? it is not in the document.

KrishnaIyer, 2024-02-26:
I think they do. We are working with Kerlink to write some documentation. I'll update this issue when that's ready.

pospospos2007, 2024-03-06:
> I think they do. We are working with Kerlink to write some documentation. I'll update this issue when that's ready.

Meantime could you share some links or files I can have reference in order to setup Basic station, don't find much resource on internet.