TheThingsIndustries/lorawan-stack-docs #1014: Document connecting Browan Pico Next Gateway to TTS

ymgupta, 2022-12-17:
#### Summary

The Browan Pico Next Gateway is a LoRa gateway with GPS, using numerous ways of connection:
Ethernet, LTE, and Wi-Fi. The current documentation doesn't have information on how to connect it to the TTS. 

#### Why do we need this?

To connect Browan Pico Next Gateway to The Things Stack. 

#### What is already there? What do you see now?

https://www.thethingsindustries.com/docs/gateways/models/

#### What is missing? What do you want to see?

Document a guide for connecting Browan Pico Next Gateway to TTS. Ref: https://www.browan.com/download/ELp/stream

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

nejraselimovic, 2022-12-17:
@NicolasMrad what should we do with these issues? I think it's not a good idea for me to pick these up as I don't have access to these devices, so I'm not able to actually test if the procedure works, hence I'm not able to write a comprehensive and accurate doc. If we have this device in the office, maybe Ben or someone else could pick it up.

NicolasMrad, 2022-12-19:
makes sense, let me check with the team if we have this device or not

KrishnaIyer, 2023-08-10:
Putting this onto the backlog until we get a device to test.

nejraselimovic, 2024-06-22:
@LDannijs do you maybe have a device on your hand to cover this?

LDannijs, 2024-06-24:
@nejraselimovic I am pretty sure we don't have one at the office, but i will check. If not i can order one.