TheThingsIndustries/lorawan-stack-docs #1016: Update video instructions for installing The Things Stack Enterprise and Open Source deployments

ymgupta, 2022-12-17:
#### Summary
The video instructions for installing The Things Stack Enterprise and open source deployments in the documentation are outdated. There are a few changes, i.e., the initialization command for the identity server database, the cockroach database has been removed, and a new component has also been introduced, i.e., Network Operations Center. Hence, we believe it would be good to have updated video instructions for Installing The Things Stack.

#### Why do we need this ?

To provide a smooth experience to the users while installing The Things Stack.

#### What is already there? What do you see now?

Video Instructions for Installing [The Things Stack Enterprise](https://www.youtube.com/watch?v=XgPSU4UkDuE) and [Open Source](https://www.youtube.com/watch?v=Owm5IUtQTx8) deployments embedded in [Installing The Things Stack](https://www.thethingsindustries.com/docs/getting-started/installation/).

#### What is missing? What do you want to see?

Updated video instructions with the latest changes included for installing The Things Stack Enterprise and Open Source deployments.

#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?

No, @benolayinka

NicolasMrad, 2023-01-18:
@benolayinka have you looked into this?

benolayinka, 2023-02-22:
Will add this to my backlog

KrishnaIyer, 2023-08-10:
This is very much nice to have. @benolayinka: Please prioritise this after the conference.

KrishnaIyer, 2023-09-29:
@benolayinka: Please pick this item up in October.