TheThingsIndustries/lorawan-stack-docs #1027: A diagram showing a message journey

NicolasMrad, 2023-01-17:
#### Summary
Have a diagram of some sort to show the journey of a lorawan message and how it travels from one component to the other.

#### Why do we need this ?
To have a better idea on the different components related to lorawan. This could be very helpful for internal and external educational use.

#### What is already there? What do you see now?
Complicated technical words

#### What is missing? What do you want to see?
A nicely presented diagram that can show the different components, abbreviations that we use and different protocols.

#### How do you propose to document this?
Diagram, PDF, slides etc. anything that would do the job. Up to @KrishnaIyer

#### Can you do this yourself and submit a Pull Request?
Can give feedback.

NicolasMrad, 2023-04-13:
@KrishnaIyer let's aim on doing this in Q2

pradeeka7, 2023-06-06:
@NicolasMrad I can take over this. I can submit a draft so @KrishnaIyer can review it.

KrishnaIyer, 2023-06-07:
Fine with me. Please go ahead.

pradeeka7, 2023-06-13:
The following are the draft images for message flows. They need to be redrawn once finalized. @KrishnaIyer can you review? any suggestions?
link to the diagram file for editing: https://docs.google.com/presentation/d/1tyR2c1R6rziEnu89MCXBG2eR51HAdDK9/edit?usp=sharing&ouid=102032046990087929579&rtpof=true&sd=true

[image]

At what point is the 'Shared Secret Key' exchanged between JS and AS?

[image]

Do we want to include message flow diagrams for handover roaming and passive roaming?

KrishnaIyer, 2023-10-01:
@pradeeka7: I'll actually pick this up myself in the backend of Q4. Thanks!