TheThingsIndustries/lorawan-stack-docs #1029: Add unable to toggle `Register network` in the Packet Broker page in the Packet Broker troubleshooting guide

ymgupta, 2023-01-21:
#### Summary

Add the information about `I'm unable to toggle **Register network** in the Packet Broker. What should I do?` in the Packet Broker troubleshooting guide. 

#### Why do we need this ?
To help users to understand the reason for behaviour and resolution to solve it.

#### What is already there? What do you see now?
https://www.thethingsindustries.com/docs/getting-started/packet-broker/troubleshooting/
#### What is missing? What do you want to see?
Add `I'm unable to toggle **Register network** in the Packet Broker. What should I do?` into the Packet Broker troubleshooting guide.

#### How do you propose to document this?

**Insights:**
Generally, to enable Packet Broker peering, The Things Stack Cloud users can enable it by toggling the `Register network` button via Console using Admin Account. It is not possible for users to enable packet broker peering if the tenant is not registered in the Packet Broker.

So, The Things Stack Cloud users can contact the TTI Support team via the `support@thethingsindustries.com` email to register the tenant in the Packet Broker. 

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

nejraselimovic, 2023-04-08:
I'm confused about this. Who does this apply to? Because [here](https://www.thethingsindustries.com/docs/getting-started/packet-broker/configure/#register-or-deregister-tenant) it says:

> To register it (tenant), toggle the Register network switch ...

So basically, if you're unable to toggle the switch, you're unable **to register the tenant**. But why are you unable to register the tenant? Why does TTI need to do it for you? @adriansmares can you pls explain?

adriansmares, 2023-04-14:
Hey - there are two issues here that make things a bit more complicated that they should be.

Fundamentally in order to receive traffic from Packet Broker, a network needs to have a set of device addresses allocated to them. We call these addresses blocks, and they are automatically allocated when people register via the normal onboarding flow (i.e. when they self register with their credit card).

Some tenants are manually created or are simply very old and do not have a device address block allocated to them. In such cases, we need to allocate them a device address block in order to allow them to receive traffic from Packet Broker. This is a manual operation.

---

There was a bug which has been fixed for version `v3.25.1` which always stopped the `Register network` switch from working for tenants which have no tenant settings whatsoever. I have fixed it in https://github.com/TheThingsIndustries/lorawan-stack/pull/3641 .

I think what we need to inform customers about is the possibility that they may need device address blocks allocated to their tenants - something like 'I have registered my network but I still do not receive traffic', which indeed should guide them towards CyberEye.

PR #1080 (2023-04-16): doc: Add support contact
#### Summary
Closes #1029 

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Style Guidelines: Documentation obeys style guidelines in [CONTRIBUTING]( [... cut]
closed 2023-04-16 by nejraselimovic