TheThingsIndustries/lorawan-stack-docs #1044: Documentation on moving devices between the applications in The Things Stack

ymgupta, 2023-02-18:
#### Summary

When moving devices from one application to another, we can use the device-claiming method. From the user's perspective, device claiming means claiming a device, i.e., Transferring Ownership. To make it more clear to TTS users, develop a document on moving devices between the applications in The Things Stack and include Device Claiming as one approach for moving devices between applications.

#### Why do we need this?

Users should be aware that how can they move devices between the applications in The Things Stack.

#### What is already there? What do you see now?

[Device Claiming](https://www.thethingsindustries.com/docs/devices/device-claiming/)

#### What is missing? What do you want to see?

Document a guide containing different approaches for moving devices between the applications in The Things Stack.

#### How do you propose to document this?
...
#### Can you do this yourself and submit a Pull Request?
No.

nejraselimovic, 2023-06-04:
I'm not sure how to proceed here. I've tried generating a TR005 QR code via ttn-lw-cli for a device registered in app1, and then scanning this code in app2, but I constantly get the `Invalid QR code data` error. I also deleted this device from app1, thinking that it cannot be added to another app if it exists in app1. This also didn't help, I get the same error. @adriansmares can you tell me if I'm doing something wrong and what can be the reason that I cannot scan the QR code I previously generated via CLI?

nejraselimovic, 2023-06-04:
I also don't see the `Claim authentication code` field mentioned [here](https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/1079) in step 5. Is this even possible using the Console? I'm also a bit confused about the claim authentication code thing. I see that when you import devices (for example using .json file), you can check the box to generate claim authentication codes for those devices, but I cannot find how to generate this code for a single device that's already registered in an app. Maybe I have an idea of how this works that's not really correlating with how it actually works :|

adriansmares, 2023-06-08:
This issue is unfortunately a bit stale. I'll give you a brief history of how this thing used to work, how it works now, and what we probably should document.

So the original device claiming mechanism, also known as the claim app (for the `/claim` website which was used for QR code scanning), was based on literary moving the end devices around. It would be strictly TTS-to-TTS, and would work by setting a claim authentication code (the missing tab that you can't find now), which would be used by the target application to move the end device.

This flow was very cumbersome and error prone - you would end up with end devices being only partially deleted from the source application, there was no UI to actually enable claiming in the source application, and in general it only worked with TTS.

The new claiming flow (which allows you to scan the QR code directly in the Console, while adding a device) is more centered around Join Server, with the idea that we don't move the end device around, but instead just tell the Join Server that from this point forward we will be the NS+AS which use a particular end device. The end device is directly registered - there is no moving around.

As a result of 'the new flow', the old flow has been deprecated and removed, as it caused more problems than it solved. 

---

Currently I would say that the best way to move end devices between applications is to use `ttn-lw-migrate` and do a v3 to v3 migration.

KrishnaIyer, 2023-08-10:
This is indeed no longer relevant since we've removed the cross-application movement functions. We will revisit documenting this when we have something for this requirement later.

closed 2023-08-10 by KrishnaIyer