TheThingsIndustries/lorawan-stack-docs #1079: Document the device claiming process with the claim authentication code via TTS Console and CLI

ymgupta, 2023-04-15:
#### Summary

[The Things Join Server](https://www.thethingsindustries.com/docs/getting-started/join-server/) is a stand-alone LoRaWAN Join Server that can be used by device makers to provision end devices to be used by all distributions of The Things Stack as well as any other standards-compliant LoRaWAN Network Servers. Users can [claim the devices](https://www.thethingsindustries.com/docs/devices/device-claiming/) using the `QR` or `Claim Authentication Code` in The Things Stack. Hence, please document the process of claiming the device using the Claim Authentication Code via TTS Console and CLI.

#### Why do we need this ?

Assist users in claiming the devices using Claim Authentication Code via The Things Stack Console and CLI.

#### What is already there? What do you see now?

Device Claiming documentation. Ref: https://www.thethingsindustries.com/docs/devices/device-claiming/

#### What is missing? What do you want to see?

Device claiming process using claim authentication code via The Things Stack Console and CLI:

**Using The Things Stack Console:**

1. In The Things Stack Console, select the appropriate application and click on the `+ Register end device` option.
2. In the `Input Method`, select the `Enter end device specifics manually` radio button.
3. Fill `Frequency plan`, `LoRaWAN version`, and `Regional Parameters version` as required.
4. In the `Provisioning information` section, fill the `JoinEUI` of users The Things Join Server (e.g., `70B3D57ED0000000`), and click on `Confirm`.
5. Fill `DevEUI` and the `Claim authentication code` of the device that needs to claim from The Things Join Server.
6. Fill `Enter device ID` and click on the `Register end device` button.

**Using The Things Stack CLI:**

1. CLI Command:
```
./ttn-lw-cli create <appliation-id> <device-id> \
--dev-eui <dev-eui> \
--app-eui <app-eui> \
--frequency-plan-id <frequency plan> \
--lorawan-version <lorawan-version> \
--lorawan-phy-version <lorawan-phy-version> \
--supports-join \
--claim-authentication-code.value <claim-authentication-code>
```
2. JSON file:
-  Create a JSON file of the device. Refer to the below example format.
```
{
  "ids": {
    "device_id": "111111111111111",
    "dev_eui": "1111111111111111",
    "join_eui": "1111111111111111"
  },
  "lorawan_version": "MAC_V1_0_3",
  "lorawan_phy_version": "PHY_V1_0_3_REV_A",
  "frequency_plan_id": "EU_863_870",
  "supports_join": true,
  "claim_authentication_code": {
    "value": "1234"
  }
}
```
- Use the below example CLI command to create the device using the JSON file.
```
 ttn-lw-cli end-device create --application-id test-application < create-device.json
```

**Note:** The cluster join server should be set as false in the CLI configuration for claiming the devices via CLI. Below is an example:

```
join-server-enabled: false
join-server-grpc-address: <tenant-id>.<eu1/eu2/au1/nam1>.cloud.thethings.industries:8884
```

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic.

nejraselimovic, 2023-04-16:
I'll wait with this until #1076 is merged

KrishnaIyer, 2023-08-10:
This is unblocked and definitely good to have. @nejraselimovic, if you have the time for this, please take this up. If not I'll pick this up at the end of Q3.

nejraselimovic, 2023-08-10:
> This is unblocked and definitely good to have. @nejraselimovic, if you have the time for this, please take this up. If not I'll pick this up at the end of Q3.

I can pick this up now

PR #1219 (2023-12-03): doc: Document claiming devices using claim auth code
#### Summary
Closes #1079 

#### Notes for Reviewers
Is this good place to locate these instructions? I find it a bit weird that it's separated from QR code-related docs and adding devices, but I somehow don't see these in Adding Devices. Also I'm curious, why does `join-server-enabled` option have to be set to false?

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_P [... cut]
closed 2023-12-03 by nejraselimovic