TheThingsIndustries/lorawan-stack-docs #1083: Release v3.25.1

adriansmares, 2023-04-18:
#### Overview

This is a checklist for releases. This is filled in by both the releaser and the reviewer where necessary.

#### Update Documentation

- [ ] Create a new release in the [`whats-new` section](https://github.com/TheThingsIndustries/lorawan-stack-docs/tree/master/doc/content/whats-new) and copy the release CHANGELOG from [TheThingsIndustries/lorawan-stack](https://github.com/TheThingsIndustries/lorawan-stack). The title is the release version and the date is the release date.
- [ ] Copy the contents of the latest UPGRADING.md from the [lorawan-stack-aws](https://github.com/TheThingsIndustries/lorawan-stack-aws) to [/getting-started/aws/ecs/changelog](https://github.com/TheThingsIndustries/lorawan-stack-docs/tree/master/doc/content/getting-started/aws/ecs/changelog/index.md)
- [ ] Remove empty sections from the created release file.
- [ ] Update the [documentation version](https://github.com/TheThingsIndustries/lorawan-stack-docs/blob/master/doc/config/_default/config.toml#L28) to match the current PATCH, if necessary (`3.${minor}.${patch}`). Note that this previously included a "v" prefix and only included a minor, but the patch must be included or repository and CLI links will break, and no "v" should be prefixed.
- [ ] To generate documentation, create a clone of [TheThingsIndustries/lorawan-stack](https://github.com/TheThingsIndustries/lorawan-stack), and **checkout the git tag of the release**.
- [ ] To generate API documentation, run the following from within the clone of [TheThingsIndustries/lorawan-stack](https://github.com/TheThingsIndustries/lorawan-stack): 

```bash
$ tools/bin/mage ttiProto:hugoData
```

- [ ] Copy generated files to `lorawan-stack-docs` by running the following commands in `lorawan-stack-docs`:

```bash
$ rsync --recursive --delete --remove-source-files ../lorawan-stack/api/ttn.lorawan.v3/ ./doc/data/api/ttn.lorawan.v3/
$ rsync --recursive --delete --remove-source-files ../lorawan-stack/api/tti.lorawan.v3/ ./doc/data/api/tti.lorawan.v3/
```

> NOTE: This assumes that the parent directory of `lorawan-stack-docs` also contains `lorawan-stack`. If not, you'll need to adjust the commands accordingly.

- [ ] To generate and export CLI documentation from within the clone of [TheThingsIndustries/lorawan-stack](https://github.com/TheThingsIndustries/lorawan-stack), first run the following command to build the CLI:

```bash
$ go build -tags tti ./cmd/tti-lw-cli
```

- [ ] Then export the CLI documentation using following commands:

```bash
$ HOME='$HOME' ./tti-lw-cli gen-md-doc -o ../lorawan-stack-docs/doc/content/ttn-lw-cli
```

```bash
$ HOME='$HOME' ./tti-lw-cli gen-json-tree -o ../lorawan-stack-docs/doc/data/commands
```

- [ ] Replace links to `end-devices/templates` with `templates` as they are not exported correctly:

```bash
$ sed -i.bak 's/end-devices_templates/templates/g' ../lorawan-stack-docs/doc/content/ttn-lw-cli/ttn-lw-cli_end-devices.md
```

- [ ] Clean up:

```bash
$ rm ../lorawan-stack-docs/doc/content/ttn-lw-cli/ttn-lw-cli_end-devices.md.bak
```

> NOTE: This assumes that the parent directory of `lorawan-stack-docs` also contains `lorawan-stack`. If not, you'll need to adjust the commands accordingly.

#### Check (for reviewers)

- [ ] A new section has been created in [`whats-new`](doc/content/whats-new) with the corresponding CHANGELOG from [TheThingsIndustries/lorawan-stack](https://github.com/TheThingsIndustries/lorawan-stack). The title is the release version and the date is the release date.
- [ ] The latest UPGRADING.md from the [lorawan-stack-aws](https://github.com/TheThingsIndustries/lorawan-stack-aws) has been copied to [/getting-started/aws/ecs/changelog](https://github.com/TheThingsIndustries/lorawan-stack-docs/tree/master/doc/content/getting-started/aws/ecs/changelog/index.md)
- [ ] The documentation version is up to date with the latest PATCH (`3.${minor}.${patch}`). No "v" is prefixed.
- [ ] The TTI and TTN API documentation has been generated and updated in [doc/data/api](https://github.com/TheThingsIndustries/lorawan-stack-docs/blob/master/doc/data/api). This includes the following files:

```
doc/data/api/tti.lorawan.v3/messages.yml
doc/data/api/tti.lorawan.v3/services.yml
doc/data/api/tti.lorawan.v3/enums.yml
doc/data/api/ttn.lorawan.v3/messages.yml
doc/data/api/ttn.lorawan.v3/services.yml
doc/data/api/ttn.lorawan.v3/enums.yml
```

- [ ] The TTI CLI documentation has been generated and updated in [doc/data](https://github.com/TheThingsIndustries/lorawan-stack-docs/blob/master/doc/data). This includes the following files:

```
doc/data/commands/ttn-lw-cli.json
doc/content/ttn-lw-cli/*.md
```

- [ ] All generated documentation matches the version that is being released.

NicolasMrad, 2023-05-05:
@nejraselimovic can we put release issues on priority please? Let's try to merge them as fast as possible

KrishnaIyer, 2023-05-05:
This whole thing should be automated. @NicolasMrad can you please create an issue and assign it to me for Q2? I'll pick this up.

nejraselimovic, 2023-05-07:
I can continue doing these until the process is automated. But I'm having issues generating these docs at the moment, not sure why

[image]

PR #1095 (2023-05-10): doc: Add v3.25.1 release docs
#### Summary
Closes #1083 

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Style Guidelines: Documentation obeys style guidelines in [CONTRIBUT [... cut]
closed 2023-05-10 by nejraselimovic