TheThingsIndustries/lorawan-stack-docs #1093: Automate the release process

NicolasMrad, 2023-05-05:
#### Summary
The release process should be automated so that it's faster

#### Why do we need this ?
Automating it will making the release process faster.

#### What is already there? What do you see now?
Currently it's being done manually.

#### What is missing? What do you want to see?
To automate the whole thing

#### How do you propose to document this?
This issue has the checklist of what is usually done in a release:
https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/1083

#### Can you do this yourself and submit a Pull Request?
No

KrishnaIyer, 2023-09-29:
@happyRip: Please prioritise this issue for October. This saves us a lot of time from doing manual work.

KrishnaIyer, 2023-12-05:
I'll pick this back up as part of the documentation work I'm working on.

closed 2024-03-19 by KrishnaIyer