TheThingsIndustries/lorawan-stack-docs #1101: Improve the API fields data type information in the API documentation

ymgupta, 2023-05-15:
#### Summary

The Things Stack users are often confused with API field data types i.e., in the documentation API field type `type: bytes` does not imply `base64`. Also, the type of API field interpreted should depends on the actual message. For example, the `frm_payload` of the [ApplicationDownlink](https://github.com/TheThingsNetwork/lorawan-stack/blob/8f5e4e1984dfcd9d9007fd2f489d89bb40cd6712/api/messages.proto#L347) is read as a `base64` type, whereas the `dev_eui` field of the [EndDeviceIdentifiers](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.25/api/identifiers.proto#L41) is read as `hex`. 

#### Why do we need this ?

To help users in distinguishing between different data types of API fields.

#### What is already there? What do you see now?

https://www.thethingsindustries.com/docs/reference/api/

#### What is missing? What do you want to see?

API fields with more details on the data types/examples.
@adriansmares , could you please add your insights here?

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

adriansmares, 2023-05-15:
We need to update our tooling in order to deal with this inconsistency. I will pick this up.

KrishnaIyer, 2023-10-01:
@adriansmares: It would be great if you could find some time to pick this up this month.

KrishnaIyer, 2023-12-05:
I'll pick this up as part of the API documentation item that I'm working on.

nejraselimovic, 2024-03-17:
hey @KrishnaIyer was this covered in https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1230?

KrishnaIyer, 2024-03-19:
I think this requires a few small updates to the generated docs. I'll pick this up later this month.