TheThingsIndustries/lorawan-stack-docs #1107: Add note point on VAT ID requirement for The Things Stack Cloud Subscription

ymgupta, 2023-05-20:
#### Summary

Generally, for commercial use cases, users should use The Things Stack Cloud Edition, which is SLA backed. Hence, for subscribing to the tenant in The Things Stack, users should provide a valid VAT ID.

However, if the user wants to use The Things Stack for non-commercial purposes and prototyping, they can use The Things Stack Community Edition, which doesn't require the VAT ID.

#### Why do we need this ?

To assist users in subscribing to The Things Stack Cloud.

#### What is already there? What do you see now?

[Subscription and Billing Management](https://www.thethingsindustries.com/docs/the-things-stack/cloud/billing-and-subscription-management/#purchasing-the-things-stack-subscription)

#### What is missing? What do you want to see?
Add the below insights on VAT ID in [Subscription and Billing Management](https://www.thethingsindustries.com/docs/the-things-stack/cloud/billing-and-subscription-management/#purchasing-the-things-stack-subscription) documentation.

**Insights:**

- If the users want to use The Things Stack non-commercial use case and prototyping, they can use [The Things Stack Community Edition](https://www.thethingsindustries.com/docs/reference/ttn/), which does not require the VAT ID.

- If the users want to use The Things Stack Cloud and don't have a valid VAT ID, they can reach out to [Sales Team](mailto:sales@thethingsindustries.com).

#### How do you propose to document this?
...

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

nejraselimovic, 2023-05-29:
Can you please tell me where one enters this VAT ID? I've tried adding a subscription for Cloud on https://accounts.thethingsindustries.com, and I get to the point of entering personal and credit card info, but it says excl. VAT and I don't see where this info should be provided. P.S. I'll do an update to whole billing and subscription management docs and cover this along

ymgupta, 2023-06-10:
@nejraselimovic,

Previously, we see the VAT ID in billing address page. Below is the screenshot attached by an user.

[image]

nejraselimovic, 2023-06-10:
Again, I don't see that field, I see it like attached on image below. Maybe it was redesigned in the meantime, or it has something to do with the account, i.e. if I'm registered as a person or a business?

[image]

nejraselimovic, 2023-10-15:
I will close this issue as I went through this process multiple times and haven't seen that **VAT ID** field anywhere. Price in the **Summary** section even says "excl. VAT". @rish-c if I'm wrong, please reopen the issue and let me know where VAT ID can be set

closed 2023-10-15 by nejraselimovic