TheThingsIndustries/lorawan-stack-docs #1135: Add extended NOC configuration in the NOC configuration options

ymgupta, 2023-07-01:
#### Summary

The Network Operations Center provides aggregated insight in a network operated with The Things Stack and it has two tiers, i.e., extended, & normal. Also, the Network Operations Center applications and end devices dashboards available with extended NOC are no longer experimental from `v3.26.0`.

The users of The Things Stack Enterprise deployments need to configure the extended NOC configurations in their `ttn-lw-stack-docker.yml` file for accessing the extended NOC dashboard. Add the extended NOC configuration option to [Network Operations Center Options](https://www.thethingsindustries.com/docs/reference/configuration/network-operations-center/) documentation. 

#### Why do we need this ?

To assist users in configuring the extended NOC in their enterprise deployment. 

#### What is already there? What do you see now?

[Network Operations Center Options](https://www.thethingsindustries.com/docs/reference/configuration/network-operations-center/) documentation.

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

**Insights:** 

Add the below extended NOC configuration option (i.e., `noc.access.extended`) to [Network Operations Center Options](https://www.thethingsindustries.com/docs/reference/configuration/network-operations-center/) documentation for enabling/disabling the extended NOC dashboards.
```
noc:
  access:
    extended: false
```

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic.

KrishnaIyer, 2024-01-26:
@ymgupta: This is just 3 lines. Please consider creating a PR yourself.

ymgupta, 2024-02-10:
The following pull request has addressed this request and the required information is available in the documentation now.
Ref: https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1184

closed 2024-02-10 by ymgupta